sarfata/pi-blaster #33: Raspberry Pi 2 Incompatibility

maxjwalker, 2015-02-12:
PI-Blaster is no longer compatible with the updated system image used since the release of the Raspberry Pi 2 http://direct.raspberrypi.org/forums/viewtopic.php?t=99115&p=690599

phueper, 2015-02-13:
According to the above link the DMA base hase moved from 0x20000000 to 0x3f000000

i tried to change that in pi-blaster.c :

-
-#define DMA_BASE               0x20007000
+#define DMA_BASE               0x3f007000
 #define DMA_LEN                        0x24
-#define PWM_BASE               0x2020C000
+#define PWM_BASE               0x3f20C000
 #define PWM_LEN                        0x28
-#define CLK_BASE               0x20101000
+#define CLK_BASE               0x3f101000
 #define CLK_LEN                        0xA8
-#define GPIO_BASE              0x20200000
+#define GPIO_BASE              0x3f200000
 #define GPIO_LEN               0x100
-#define PCM_BASE               0x20203000
+#define PCM_BASE               0x3f203000
 #define PCM_LEN                        0x24

recompiled, but GPIOs / PWM still does not seem to work

phueper, 2015-02-13:
maybe also worth to follow this: https://github.com/richardghirst/PiBits/issues/43

phueper, 2015-02-23:
richardghirst seems to have fixed servoblaster (see comment in the above bug), he is now using mailbox commands to request memory and some more changes as far as i can see...  i am not sure how easy it is to adapt to pi-blaster...  the mailbox files look like they are open source from Broadcom so the could be copied to pi-blaster as well... then the different membases would need to be used in pi-blaster i guess...

popoklopsi, 2015-02-23:
You just can use servoblaster, there is no need more for pi-blaster

Keridos, 2015-02-23:
Servoblaster is the original repository  this is a fork of as far as I know and it had no commit in the past 12 months, so how should that be more up to date than this?

phueper, 2015-02-23:
In case anyone cares... i started adapting richardghirsts above changes mentioned above from http://www.raspberrypi.org/forums/viewtopic.php?f=28&t=99115&p=699651#p699651 to pi-blaster... 
not quite ready for a pull request yet (printf debugging ;)) but the diff is here:
 https://github.com/phueper/pi-blaster/compare/sarfata:master...master

I just got my first pi-blaster PWM signal on my Pi 2 using that code... it did not seem very stable but that may be due to me adding -g to CFLAGS and maybe other problems... 

I did not try the code on my Pi 1 B+ either... will do so in the next days and maybe post updates.

popoklopsi, 2015-02-23:
Pi-Blaster was created because servoblaster coulnd't get 100% percent of the power. But meanwhile it can.

See here: http://makezine.com/2013/02/12/pi-blaster-improves-software-pwm-on-raspberry-pi/

"While exploring his options with software PWM, he came across ServoBlaster, which cleverly offloads the task of switching the pins to the direct memory access (DMA) controller. Since ServoBlaster was meant for controlling hobby servo motors using pulses, it didn’t give Thomas the full 0 to 100% PWM control he sought"

phueper, 2015-02-24:
I tested my current state on my Pi 1 B+ and Pi 2... it is generating a PWM Signal on both, however i can currently see 2 issues:

- on Pi 1 the Frequency is correct at 100 Hz on my scope, on the Pi 2 however, the cycles are much shorter and the frequency on the scope is shown as 1.7 kHz
- the code is not restartable, when i have started pi-blaster already it seems to hang in mem_alloc on the second start (at least on Pi 2, not verified on Pi 1 yet)

So... still some work left ;)

sarfata, 2015-03-03:
@phueper Thanks for working on this and sharing the result.

@Keridos This code has seen little commits because we did not need them ;) I do not have a Raspberry Pi 2 so I have not been able to try this out yet. I would be happy to review and merge a pull-request when @phueper gets it working.

phueper, 2015-03-03:
i didn't have a lot of time lately... still working on it though... and having fun :) my guess is that something is wrong in the setup of the PWM/CLK/DMA Registers but i haven't figured it out... with servoblaster the frequency on Pi2 is ok so i must be missing something... i'll keep you updated, maybe i'll find some time this weekend

wholelottajam, 2015-03-18:
@phueper Any news on the Pi 2 issue?

kboyington, 2015-03-19:
Any news? I just got a PI2 and am looking forward to this!

phueper, 2015-03-19:
Sorry... family kept me busy the last few weeks... I am planning to work on
it this weekend... from what i found so far the clocking setup seems to be
ok, i suspect something wrong in the DMA engine causing the frequency to be
too high... but i havent found it yet... the current state in my fork is
still the most up-to-date code i have :(

pfnegrini, 2015-03-20:
Really appreciate your efforts and hope you will find some time to fix it: the life of many robots depend on this :)

kboyington, 2015-03-22:
@phueper Thanks for the update! Enjoy the weekend! Looking forward to the next commit!

phueper, 2015-03-24:
Still no good news, sorry...

The problem still is that the frequency of the PWM is too high (~1kHz) the
Duty Cycle is correct so my current suspiciion is that the DMA Control
Blocks are setup ok, but something with the clocking setup does not work...
since it did work in the PI1 last time i checked i probably have some
offset for the clk/pwm registers wrong but i cant figure it out...

I will checkin my current code into my repo with lots of debugging in case
someone wants to take a look and will continue to investigate.

Regards, Pattrick

phueper, 2015-03-26:
Hi,

FINALLY... some good news... i think i have figured out the problem i had... wrong calculation of the wait trigger address for the DMA... i need to cleanup the code a little bit... probably will checkin a version sometime later today 

Regards, Pattrick

phueper, 2015-03-26:
I committed the fix into https://github.com/phueper/pi-blaster 

The code only worked by chance on the Pi1 since the Pi1, the calculation of the phys_fifo_addr was wrong.

I am not quite ready for a Pull Request, some debug printfs need to be removed and i would also like to test the code on my Pi1... but if anyone wants to try the current state it should be working on Pi2 now... any feedback is appreciated

Regards, Pattrick

wholelottajam, 2015-03-26:
Thank you very much. I will test it on Pi2 as soon as possible.

phueper, 2015-03-27:
Quick Test on my Pi1 indicates that it is also still working on Pi1... so here comes the Pull Request

radjkarl, 2015-03-28:
Hi,
thank you very much for this great package!

I found 2 issues playing with phueper's fork:

1. I have a python program to  control 5 leds via PWM
- to access them i execute 'echo [port]=[val]...'
- at the end all ports released
however, after  executing the program 2 times the leds remain dark
if i start ./pi-blaster i get the following output:
--
Failed to create mailbox device
pi-blaster stopped
--

2. At the end of the boot process i start the xserver with a Qt-based application.
If pi-blaster is installed the desktop remains black.
As soon as I uninstall pi-blaster I get the Qt-Application back.

My ugly workaround for both problems is at the moment to install and uninstall pi-blaster (sudo make (un)install) everytime i start my application.  

Cheers,
Karl

phueper, 2015-03-28:
>    2. to access them i execute 'echo [port]=[val]...'

Hi,

Thanks for testing...

The "failed to create mailbox device" message is maybe because you didn't
start pi blaster with sudo?

I have not tried running with X,  but others have reported problems with
pi-blaster and X as well... Will try to investigate.

Cheers, Pattrick

phueper, 2015-03-31:
I can confirm that X is not starting (screen remains dark) when pi-blaster is running... if i start pi-blaster after X everything seems to be OK. 
I have no idea why this happens :(

phueper, 2015-03-31:
... i think i have an idea what the problem might be... expect an update today, maybe tomorrow...

phueper, 2015-03-31:
PR #38 updated... works for me now with startx while pi-blaster is running

PR #38 (2015-04-15): Make pi-blaster work on Pi2
fixes #33 

The basic logic was merged from @richardghirst s ServoBlaster fix for Pi2 but adapted where it seemed nicer/more clear to me...

- use mbox commands to allocate and then mmap memory areas for the DMA ControlBlocks and Sample Data
- add mbox command to read the firmware/board revision
- detect the board model from the mbox data, not by reading /proc/cpu (this is different from the ServoBlaster code but to me seems cleaner and more sustainable for future models)
- since Pi1 and Pi2 use different base adresses for mapping peripherals (DMA, PWM, PCM, CLK reg [... cut]
closed 2015-04-15 by sarfata