sarfata/pi-blaster #35: Systemd support

bondar-pavel, 2015-03-04:
I saw several commits that are expected to enable systemd support for pi-blaster.
However it did not worked out for me out-of-box.
Observed a few errors and it is still tries to use init.d dir.
Here is the installation log:
```
[apostal@alarmpi pi-blaster]$ ./autogen.sh
Generating configure files... may take a while.
configure.ac:28: installing './compile'
configure.ac:8: installing './install-sh'
configure.ac:8: installing './missing'
Makefile.am: installing './depcomp'
Preparing was successful if there was no error messages above.
Now type:
  ./configure && make
Run './configure --help' for more information
[apostal@alarmpi pi-blaster]$ ./configure && make
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
./configure: line 2415: PKG_PROG_PKG_CONFIG: command not found
./configure: line 2421: --variable=systemdsystemunitdir: command not found
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating config.h
config.status: executing depfiles commands
make  all-am
make[1]: Entering directory '/home/apostal/git/pi-blaster'
gcc -DHAVE_CONFIG_H -I.     -Wall -pedantic -MT pi-blaster.o -MD -MP -MF .deps/p                                                                                        i-blaster.Tpo -c -o pi-blaster.o pi-blaster.c
pi-blaster.c:39:1: warning: C++ style comments are not allowed in ISO C90
 // Created new known_pins with raspberry pi list of pins
 ^
pi-blaster.c:39:1: warning: (this will be reported only once per input file)
pi-blaster.c: In function âudelayâ:
pi-blaster.c:203:9: warning: initializer element is not computable at load time
  struct timespec ts = { 0, us * 1000 };
         ^
pi-blaster.c: In function âgo_go_goâ:
pi-blaster.c:641:2: warning: ISO C90 forbids mixed declarations and code [-Wpeda                                                                                        ntic]
  char *lineptr = NULL, nl;
  ^
mv -f .deps/pi-blaster.Tpo .deps/pi-blaster.Po
gcc  -Wall -pedantic   -o pi-blaster pi-blaster.o
make[1]: Leaving directory '/home/apostal/git/pi-blaster'
[apostal@alarmpi pi-blaster]$ sudo make install
make[1]: Entering directory '/home/apostal/git/pi-blaster'
 /usr/bin/mkdir -p '/usr/sbin'
  /usr/bin/install -c pi-blaster '/usr/sbin'
 /usr/bin/mkdir -p '/etc/init.d'
 /usr/bin/install -c pi-blaster.boot.sh '/etc/init.d'
make  install-data-hook
make[2]: Entering directory '/home/apostal/git/pi-blaster'
update-rc.d pi-blaster.boot.sh defaults
make[2]: update-rc.d: Command not found
Makefile:846: recipe for target 'install-data-hook' failed
make[2]: *** [install-data-hook] Error 127
make[2]: Leaving directory '/home/apostal/git/pi-blaster'
Makefile:770: recipe for target 'install-data-am' failed
make[1]: *** [install-data-am] Error 2
make[1]: Leaving directory '/home/apostal/git/pi-blaster'
Makefile:720: recipe for target 'install-am' failed
make: *** [install-am] Error 2
```

patlux, 2015-05-24:
Hi,

I updated the aur package for archlinux users:
https://aur.archlinux.org/packages/pi-blaster-git/

You can easily install it with yaourt 
```
yaourt -S pi-blaster-git
```
and activate the systemd-service with
```
sudo systemctl enable pi-blaster
```

sarfata, 2015-05-25:
@patlux That is great news. Could you make a PR to add this information in the `README.md` file?

sarfata, 2016-04-19:
@patlux Thanks again for this. If we merge #53, will we break your package? It looks like it relies on make install right now.

Does Arch support systemd files?

sarfata, 2016-04-19:
added archlinux instruction.

closed 2016-04-19 by sarfata