sarfata/pi-blaster #42: RaspberryPi crash when writing new pin state to often

felicisflame, 2015-05-26:
Hi.
I am using a Raspberry Pi 2 Model 5 and when I write the pin state to often (e.g. in a python or java script (see below))  the the pi crashes and reboots without any message.
Code extract(Python):
for i in range(10):
    file=open("/dev/pi-blaster", "w")
    file.write("17=0.1" + "\n")
    file.close()
    sleep(1)
    file=open("/dev/pi-blaster", "w")
    file.write("17=0.05" + "\n")
    file.close()
    sleep(1)
In this example my pi crashes somewhere around the seventh or eighth pass.
Do you have any idea what I could do?

teh1, 2015-06-16:
Confirmed that this happens on this my end.

Bump!

sarfata, 2015-06-23:
@felicisflame why did you close this? Did you find a fix?

teh1, 2015-06-23:
Still happens for me. Maybe we should re-open it?

sarfata, 2015-06-23:
Yes. I don't have a solution but I think we need to keep this open to track this problem and hopefully fix it someday.

felicisflame, 2015-06-24:
Oh sorry for closing the issue. I didn't get that there was somebody with this problem too. In my case I found a solution some days ago:
I was controlling a very small servo with the program (as it is ment to) and whenever I wrote a new state, my servo produced a voltage drop. Due to the fact that the servo's power supply was the pi, the pi didn't like this drop and it rebooted every time. The problem occured from time to time and the frequent changes of the pin state only increased the chance for causing the reboot. So it's not pi-blaster's fault but caused by the circuit.
In my case it was enough to connect the servo to an external power supply.
If my explanation was not understandable enough, feel free to ask me more detailed questions.

teh1, 2015-06-24:
Ah! Makes perfect sense. For my testing purposes I have been using the pi as a power source. I guess when I wire everything up in its final state I won't experience this.

Thanks for explaining.

YemSalat, 2015-07-22:
Hey guys, the issue is with your wiring, not with the library.

I guess you are using the same power supply for your Pi and for whatever it is that you're controlling. Your device is generating voltage spikes which effectively reboot your Pi.

sarfata, 2015-08-09:
Closing now since this seems resolved.

closed 2015-08-09 by sarfata