sarfata/pi-blaster #47: Can't open device file

tgrosinger, 2015-08-17:
I have built and installed Pi-Blaster on my Raspberry Pi B running AlarmPi. Attempting to run (as root) results in this error message:

```
Can't open device file: /dev/pi-blaster-mbox
Try creating a device file with: sudo mknod /dev/pi-blaster-mbox c 100 0
```

I have tried running the suggested command, and even deleted that file and re-run the command again. Unfortunately I just get that message over and over.

Any help on what I am missing or doing wrong? Thanks.

tgrosinger, 2015-08-17:
I ran a full new build from scratch just to make sure I didn't miss a step. Similar results:

```
[tony@alarmpi pi-blaster]$ ./pi-blaster 
Failed to create mailbox device
pi-blaster stopped.

[tony@alarmpi pi-blaster]$ sudo ./pi-blaster
Can't open device file: /dev/pi-blaster-mbox
Try creating a device file with: sudo mknod /dev/pi-blaster-mbox c 100 0
```

blydro, 2015-08-19:
I have the same problem. Also running Arch Linux for Raspberry Pi -- perhaps the problem lies there?

anzas, 2015-08-19:
Same problem here as well. I'm using Debian. It was working at first, but stopped after I installed OS  updates (including new kernel). I checked that the device file /dev/pi-blaster-mbox exists.

Linux raspberrypi 4.1.4+ #808 PREEMPT Thu Aug 6 15:51:03 BST 2015 armv6l GNU/Linux
Distributor ID: Debian
Description:    Debian GNU/Linux 7.8 (wheezy)
Release:        7.8
Codename:       wheezy

sarfata, 2015-08-20:
Can one of you guys update the code, re-compile and share the output?

I have added a one-line change to display the error we get from the system. (cf 74510ff5da4e61d0b6f31837757116eb2951ca7d)

blydro, 2015-08-20:
Now I'm thoroughly confused...

```
$ sudo ./pi-blaster
Can't open device file: /dev/pi-blaster-mbox
No such device or address
```

but I'm pretty sure it does exist:
```
$ ls -l /dev/pi-blaster-mbox 
crw------- 1 root root 100, 0 Aug 20 08:58 /dev/pi-blaster-mbox
```

I've even run the program several times, but get the same error every time.

anzas, 2015-08-20:
According to this https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=117547
the device number for mailbox (/dev/vcio) should be 249 instead of 100 in the 4.1 kernel.

I did the change to mailbox.h (#define MAJOR_NUM 249)
Now I get this and a hangup at the end
```
update-rc.d pi-blaster.boot.sh defaults
update-rc.d: using dependency based boot sequencing
/etc/init.d/pi-blaster.boot.sh start
ioctl_set_msg failed:-1
MBox Board Revision: 0
ioctl_set_msg failed:-1
DMA Channels Info: 0, using DMA Channel: 14
Using hardware:                   PWM
Number of channels:                 9
PWM frequency:                 100 Hz
PWM steps:                       1000
Maximum period (100  %):      10000us
Minimum period (0.100%):         10us
DMA Base:                  0x20007000
ioctl_set_msg failed:-1
```
Guess something else needs to be done as well.

knusperriegel, 2015-09-07:
I got the same issue when trying to setup a jessie based pi1.

=> As long as this is not yet fixed: Where can I download the old (last not "/dev/pi-blaster-mbox"-based) pi-blaster master.zip?

SIPHOR3ON, 2015-09-10:
Hi,
got the same issue here. This one helped:
http://www.homofaciens.de/download/pi-blaster-master.zip

Hope I could help you

larnera, 2015-09-13:
Hi Siphor3on. I got the same error message and so I've downloaded the file you've hosted and extracted it onto my pi2. I've ran ./autogen.sh and I get bash: ./autogen.sh: perrmission denied. 

Any idea what I'm doing wrong? as you can tell, I'm pretty new and I'm desperate to get my servos working on my pi2. 

Cheers. 
[image]

improti, 2015-09-13:
@sarfata I did as you asked. Using the latest version my output on a Pi2 running Jessie is:

```
root@pi-home:/usr/local/src/pi-blaster# ./pi-blaster 
Can't open device file: /dev/pi-blaster-mbox
No such device or address
```
Doesn't sound very helpful though? The file does exist btw.:
```
root@pi-home:/usr/local/src/pi-blaster# ls -lsah /dev/pi-blaster-mbox 
0 crw------- 1 root root 100, 0 Sep 13 21:19 /dev/pi-blaster-mbox
```

zador-blood-stained, 2015-09-14:
The thing with newer kernels, I believe, is to use /dev/vcio device node, provided by the kernel itself, rather then creating one. I made a pull request #49 that shoud fix this issue. Please try out my changes

larnera, 2015-09-14:
The fix worked. Servo is working great !

sarfata, 2015-09-14:
closed by #49

closed 2015-09-14 by sarfata

born2net, 2015-10-02:
this fix needs to be put into final release, it works great!

sarfata, 2015-10-10:
@born2net The fix has been merged. What did you mean "into final release"?

improti, 2015-10-11:
Works for me too, at least on the Raspi2 (still have to check the older ones). But thanx a bunch mate!