sarfata/pi-blaster #50: Invalid channel number 4

yumengwang03, 2015-09-23:
I cloned pi-blaster folder to my home directory on the pi and then installed pi-blaster.js. When I typed in sudo ./pi-blaster, it tells me that there are 8 channels. But then, when I used "echo "4=0.5">/dev/pi-blaster", it showed me "invalid channel number 4". Also tried channel 17 and got the same result. What might be the reasons?

Thank you!

improti, 2015-10-23:
Just to clarify ... The number 4 in your command `echo "4=0.5" > /dev/pi-blaster` addresses GPIO pin number 4. My impression was the number of channels is the number of pins you can control concurrently. I am not sure about the latter, though. And I cannot help you with your problem of the invalid channel number, either. But you shouldn't just put numbers in there, at least not if you got other stuff connected to those pins. You might destroy some hardware otherwise ...

sarfata, 2015-12-29:
@improti is right. The number of channels is equal to the number of pins.

You should not be getting "Invalid channel number" with the line that you are sending ([source](https://github.com/sarfata/pi-blaster/blob/master/pi-blaster.c#L832)). Are you still getting this error? Sorry about the delay looking into this ...

sarfata, 2016-04-19:
Closing. Please re-open if this is still an issue for you and tell us what hardware you are testing on.

closed 2016-04-19 by sarfata