sarfata/pi-blaster #51: How to stop LED from blinking

dhunink, 2015-09-27:
Hi There,
I connected a RGB LED strip to my raspberry and am trying to control the Red line using the awsome Pi-Blaster. Unfortunally my LED strip is not being stable: it keeps looping from on to off. I posted a little demo on https://www.youtube.com/watch?v=qEt5gFcJue4. 

The behaviour seen in the link above is the direct result of 
    echo "1=0" > /dev/pi-blaster

It is driving my totally mad; I can not figure out why this is happening. Any thoughts are very appriciated!

sarfata, 2015-09-29:
What type of rgb strip do you have?

dhunink, 2015-09-29:
I bought 2 of this and linked them: http://www.ledstripkoning.nl/led-strips-rgb/losse-ledstrips-rgb/5-meter-150-leds-rgb-led-strip/
So we have 10meter LED strip running 24V, LED type 5050 SMD. 
I can contact the seller to ask for more details if needed.

zador-blood-stained, 2015-09-29:
What power supply do you use to power your LED strip? Did you connect RPi output to MOSFET, bipolar transistor or some IC to control LED strip brightness?

dhunink, 2015-09-30:
We use a 24V/10A power-supply (model S-250-24) and connected the LED to MOSFET TIP120.
I've added some pictures, including more details on the  LED strip.

The power supply:
[image]

The connected MOSFET:
[image]

The MOSFET in more detail:
[image]

The LED Strip:
[image]

zador-blood-stained, 2015-09-30:
Did you use a current limiting resistor between RPi pin and transistor's base? BTW, TIP120 is not a MOSFET, it is a Darlington transistor (array).

dhunink, 2015-09-30:
No I did not, only the TIP120 (which is indeed not a MOSFET, my mistake, sorry). Really the only thing between the LED's and the RPi is the TIP120 (one for each color)

zador-blood-stained, 2015-09-30:
Resistors are **required** in this case, using GPIOs without limiting current may damage RPis SoC.
I would recommend using 600 Ohm to 1 kOhm resistor between GPIO and transistors base.
Then I would test LEDs without PWM by writing logical 1 to appropriate pin (i.e. with `gpio` command from wiringPi, keep in mind that it is using different pin numbering system by default, see `man gpio` for details).
If LEDs do not blink in this case and transistors do not quickly heat up, then I would proceed to testing PWM with pi-blaster.

zador-blood-stained, 2015-09-30:
I assume that your wiring looks (will look) like this:
[image]

dhunink, 2015-09-30:
Thanks for all your time and efforts, highly appreciated!
Our setup features a TIP120 with 3 pens, instead of the 8 pins you drawn. Apart from the scheme is absolutely correct!

zador-blood-stained, 2015-09-30:
You're welcpme.
This is just a weird looking symbol in KiCad library for this transistor. Indeed it is in 3 pin TO-220 package.

dhunink, 2015-09-30:
Alright :) We're going to test with the 620 (or 1K) asap, although this may take a couple of days before we are able to do so. Definitely coming back to you with the results!

improti, 2015-10-08:
Although this is all way off-topic for here ... I tried to use TIP120 too, but although trying different resistors, I couldn't get the stripe to become bright. I read that using MOSFETs seems to have benefits: less temperature (and according loss), more brightness instead + you do not need the resistors as MOSFETs are controlled by voltage, not current. I currently use P16NF06L successfully on different installations. In case you got some MOSFETs at hand, you might want to try them.

zador-blood-stained, 2015-10-08:
Yes, good logic-level MOSFETs will be better than TIP120, but you still need to use resistors. MOSFETs are controlled by charging/discharging gate-channel capacitance, and without resistor you will get current spike every on->off and off->on transition. And more "powerful" MOSFET is - more capacitance - more time charging/discharging current will be exceeding maximum recommended values for RPi. TL;DR Using MOSFETs without resistors will probably work, but it is not recommended.

improti, 2015-10-08:
Interesting. I have been asking around for this and keep getting different answers. While no one really said the resistor would harm (advise was for very low Ohm resistors) they also were seen as unneeded and rather for cases where you do something wrong when connecting (or connecting a broken part). If you really are into this, can you point me to according detailed explanations maybe? Would really like to understand for what the resistor is needed and how to calculate it's dimension for that purpose.

Btw. and while we are at it. I was advised that for MOSFETs a resistor between Gate and Source would make sense to ensure the circuit is closed while the Pi is powered off. No calculation provided but 100k was given as a value. You got an opinion on that one or a hint on where to read the glory details?

zador-blood-stained, 2015-10-08:
About current limiting resistors - [these](https://electronics.stackexchange.com/questions/135133/calculate-mosfet-gate-resistor) answers look good to me. To partly correct myself - you will be okay without current limiting if you're driving something small like 1N7000 or BS170, but for MOSFETs rated for more than 10A (and especially more than 100A) source current you better use them.
About gate to source resistors - yes, if you use separate power supply for something that controls transistors gate (RPi, MCU, ...), you want a resistor between gate and source to prevent transistor from opening due to i.e. static charge. Try watching [this](https://youtu.be/gloikp9t2dA?t=78) video (or search for other articles and videos about testing MOSFET with multimeter) and if you understand why transistor is open while nothing is connected to its gate, then you can imagine that it may open accidentally if practically floating gate gets charged above threshold voltage and there will be nothing (apart from leak current, which is very small) to discharge it.

improti, 2015-10-10:
Thx mate, that's a good link. Guess I will put those resistors in after all. The MOSFETs I use are rated pretty high, which seems to bring the benefit of them not getting warm at all. It was pointed out to me, though, that with the upper limit of V(threshold) being 2,5 V they wouldn't necessary work with the 3,3 V of the Pi (one would need double the threshold to be sure?). They do for me, though. The FDN327N specs (http://www.farnell.com/datasheets/89384.pdf) look like those would be much better suited with a maximum Gate threshold of 1,5 V (even doubled still below the 3,3 V of the Pi), but those seem to be SMD, which I cannot solder :( Anyways, your effort is much appreciated, thx a bunch!

zador-blood-stained, 2015-10-10:
Gate threshold is a voltage when transistor *starts* to open, you better check datasheet graphs like Rds(Vgs) for FDN327n:
[image]
Or output characteristics for STP16NF06L:
[image]
Gate voltage of 3v is enough for 10A source current for STP.

improti, 2015-10-10:
Mate, you are great! I did understand the threshold is the lower bound under which the MOSFET is guaranteed to not switch. However the upper threshold, i.e. the upper bound beyond which the MOSFET is guaranteed to switch, seems not be specified. I guessed that is because of dependencies from various conditions but didn't get far enough with my understanding ... 

I did look at the graph of the STP before. My problem was, that nobody explained me how to read them. I think I got it now, though. And it is pretty great, because now I think I know they will work so I can keep using them. They are a lot cheaper than the alternatives that I was pointed at so far, so I'm pretty happy with that. Again: Thx a bunch!

closed 2016-04-19 by sarfata