sarfata/pi-blaster #52: Strange behaviour with concurrent pi-blaster and rpi-gpio

improti, 2015-10-11:
I'm controlling rgb led stripes using pi-blaster while running rpi-gpio in parallel to query sensors and set single leds. For sensors I use callbacks, so I do not have to query them all the time and instead get an interrupt whenever something happens. Like this:
```
    GPIO.add_event_detect(in_pin, GPIO.BOTH, callback=self._interrupt_pir)
```
For the stripes I do some fading by using this construct:
```
    fade_steps = 50
    sleep_time = 0.02
    for loopcounter in range (1, fade_steps, 1):
      [...] calculate new colour
      self.setrgb(colour_new)
      time.sleep(sleep_time)
```
Here setrgb will execute an according `echo x=y >> /dev/pi-blaster` for each of the channels.

I had to realise that each time I do a fading, the set callback stops working? For now I use a workaround like this after each fade (the differences are due to the call being made from a different class, it is working that way ...):
```
        GPIO.remove_event_detect(__GPIO_PIN_PIR)
        GPIO.add_event_detect(__GPIO_PIN_PIR, GPIO.BOTH, callback=my_pir._interrupt_pir)
```              
I assume this is not "working as intended" and therefore looking for hints to "what gives". Anyone else experienced this? Any explanations/solutions around?

improti, 2016-01-02:
Any clue on this please? It really should be possible to do both without interference, right? Anything I can check/try? Want to add that the Pi is additionally using I2C.

improti, 2016-01-11:
Ok, I think I figured this one out. I was issueing the commands to piblaster using ```os.system("/usr/bin/echo 23=0.7 >> /dev/pi-blaster")``` which immediately breaks an existing edge-detection. This isn't about piblaster at all, even just echoing a string without the redirection breaks the edge-detection. Switching to the following resolved this behaviour:
```
import subprocess
[...]
returned_status = subprocess.call(self.__CMD_ECHO + " 23=0.7 >> /dev/pi-blaster", shell=True)
```
Putting this here in case someone else runs into the same problem. See https://docs.python.org/2/library/subprocess.html#subprocess-replacements for various hints on how to use "newer" stuff ...

closed 2016-01-11 by improti

sarfata, 2016-01-11:
Thanks for following up on this. I had started looking at rpi-gpio source but it is a massive piece of code and had no idea where the two could interfere.