sarfata/pi-blaster #57: pi-blaster does not relaunched automatically on reboot

bchr02, 2015-12-23:
The README.md states:
>To start pi-blaster and have it relaunched automatically on every reboot:
sudo make install

but after doing ```sudo make install``` it does not start the pi-blaster service or start it on reboot. I have to manually start it by typing ```sudo systemctl start pi-blaster```

hosiet, 2015-12-23:
By reading current `Makefile.am`, you will find that the documentation is outdated and autostart is not true for systems using systemd. You must run `sudo systemctl enable pi-blaster.service` manually after installation.

bchr02, 2015-12-23:
Thank you! I will try this.

bchr02, 2015-12-28:
@hosiet  This worked. Thank you. You mentioned the docs were outdated, is pi-blaster not being maintained anymore?

sarfata, 2015-12-28:
Yes it is still maintained!

If one of you could propose a pull-request to fix this issue, I would gladly merge it. I think the best way to fix this might be to update the Makefile.am to do what the README says (configure program to autostart).

bchr02, 2015-12-31:
@sarfata Unfortunately, I am not familiar with how Makefile.am works. @hosiet any chance you can do this?

is it just a matter of changing:
````
systemdsystemunit_DATA = pi-blaster.service
````

to:
````
systemdsystemunit_DATA = enable pi-blaster.service
````

sarfata, 2015-12-31:
I think it's going to be more complicated. #53 is also doing a bunch of related changes so we should wait until it lands before fixing this.

sarfata, 2016-04-19:
closed by #64

closed 2016-04-19 by sarfata