sarfata/pi-blaster #58: pi-blaster restart not function properly

wanjungao, 2016-02-29:
It seems that the folder /dev/pi-blaster is created on boot. But it does not work properly.

e.g.

`
sudo echo "10=1" > /dev/pi-blaster 
`

will return immediately, even the pin is not in the configured list.

then if we do 
`
sudo pi-blaster
`

I got 

MBox Board Revision: 0xa21041
DMA Channels Info: 0x7f35, using DMA Channel: 14
Using hardware:                   PWM
Number of channels:                 3
PWM frequency:                1000 Hz
PWM steps:                        100
Maximum period (100  %):       1000us
Minimum period (1.000%):         10us
DMA Base:                  0x3f007000
Initialisation finished, pi-blaster now running as daemon, waiting for input on /dev/pi-blaster

then I did

`
sudo echo "10=1" > /dev/pi-blaster
`

And I got proper response

Not a known pin for pi-blaster

sarfata, 2016-02-29:
Can you reboot your computer and run `ps auxwww |grep pi-blaster` to see if it's actually running at boot or not?

If not can you tell us what distribution (and what version) you are running? thanks

wanjungao, 2016-03-01:
Yes. It is actually running.

But until I manually run sudo pi-blaster, the command "echo "10=0" > /dev/pi-blaster" would not prompt "Not a known pin for pi-blaster"

I changed pins like 

static uint8_t known_pins[] = {
        18,     //pin 12 fan
        16,     //pin 36 circulation pump 
        20      //pin38 PwmThemoIO
};

I am using Jessie lite on Pi2.

The more problem is that even I manually run "pi-blaster", 

`
MBox Board Revision: 0xa21041
DMA Channels Info: 0x7f35, using DMA Channel: 14
Using hardware:                   PWM
Number of channels:                 3
PWM frequency:                1000 Hz
PWM steps:                        100
Maximum period (100  %):       1000us
Minimum period (1.000%):         10us
DMA Base:                  0x3f007000
Initialisation finished, pi-blaster now running as daemon, waiting for input on /dev/pi-blaster
`

and wrong pin will prompt alert, the correct pin does not output pwm.

wanjungao, 2016-03-05:
changed to use original frequency, and problem went away.

closed 2016-03-05 by wanjungao