sarfata/pi-blaster #59: Working with Raspberry Pi3

bastardop, 2016-03-13:
Hi,

i just tried to replace my old Raspberry Pi 1B with my new Pi3 but it seems that Pi-Blaster is not working with Pi3. 
Installation worked fine but when trying to set a Pin to 1 nothing happens on the Pin

Can someone say something about using pi-blaster on pi3 or what you need from me?

Cheers

sarfata, 2016-03-13:
I have ordered one but I do not have it yet. Let's keep this issue open to track progress on this.

bastardop, 2016-03-13:
that would be really great :) as i said if you need some information or output just say a word ;)

theoatvecna, 2016-03-15:
+1 on the raspi3 fix.  This is a very useful tool and I would like to use it on the pi3.  If you can't get your hands on one soon, let me know and I will send you mine! Thanks for the great software!

mdumont, 2016-03-18:
If you need a Pi3 to work on this I'm local in the bay area and have a spare I could lend until yours arrives.

sarfata, 2016-03-18:
It is scheduled to be delivered today so I think I will be good with that. It might take me a little while to close this issue though :) but I am glad to see so many people interested!

TuxfeatMac, 2016-03-26:
First of all I also want to thank you for the great software! It would be really great to see pi-blaster also running on RPi3. I have a RPi3 up and running. If you need any support for a test run feel free to contact me. I'm not a professional software developer but a long time Linux user who knows a little more than the basics. :)

watkinspd, 2016-04-03:
+1 on seeing this work for Pi3

Also for seeing /dev/pi-blaster mapped as a device into docker containers as per the way hypriot has enabled GPIO -- example here http://blog.hypriot.com/post/lets-get-physical/

The idea would be for pi-blaster.js to work in docker containers by talking to /dev/pi-blaster on the host computer

sarfata, 2016-04-05:
I got my Raspberry Pi 3 and started looking for documentation. Apparently Broadcom has not released a datasheet yet for the Rpi3. Having it makes it a lot easier to understand where the DMA controller and the GPIO pins are mapped in the memory of this Raspberry Pi. Alternatively, I will start exploring the kernel source. It should be in there too.

The [rpitx](https://github.com/F5OEO/rpitx/) project also uses the DMA controller and has [a similar issue](https://github.com/F5OEO/rpitx/issues/14).

sarfata, 2016-04-05:
@watkinspd I love the idea of Docker on the Raspberry Pi. Thanks for the link to Hypriot. The trick is interesting but giving general access to /dev/mem is really a bit overkill and completely exposes your machine to security problems - or just buggy code that would start writing everywhere.

Could you open a separate issue to track running pi-blaster in Docker? We could discuss that more over there.

penfold42, 2016-04-18:
#62

sarfata, 2016-05-02:
closed by #62

closed 2016-05-02 by sarfata

stozk, 2016-08-02:
I have trouble getting it to work on my Pi3. I've built from source and used various fixes in the issue tracker here to make all errors disappear. I don't get any reaction. Using a Python script to verify the general function of my servo is successful, but pi-blaster, respectively pi-blaster.js doesn't produce any reaction.

Everything seems to be correctly in place:

> pi@raspberrypi:~/servo $ sudo service pi-blaster status
● pi-blaster.service - Daemon for PWM control of the Raspberry Pi GPIO pins
   Loaded: loaded (/lib/systemd/system/pi-blaster.service; enabled)
   Active: active (running) since Tue 2016-08-02 21:31:18 CEST; 21min ago
  Process: 383 ExecStart=/usr/sbin/pi-blaster $DAEMON_OPTS (code=exited, status=0/SUCCESS)
 Main PID: 390 (pi-blaster)
   CGroup: /system.slice/pi-blaster.service
           └─390 /usr/sbin/pi-blaster
Aug 02 21:31:18 raspberrypi systemd[1]: Starting Daemon for PWM control of the Raspberry Pi GPIO pins...
Aug 02 21:31:18 raspberrypi systemd[1]: Started Daemon for PWM control of the Raspberry Pi GPIO pins.