sarfata/pi-blaster #61: Docker and pi-blaster: access from within containers

watkinspd, 2016-04-06:
Would be great if there was a way to access pi-blaster from within docker container apps.

Current examples of access to GPIO involve mapping devices or volumes into the docker container during the docker run command.
http://blog.hypriot.com/post/lets-get-physical/
https://hub.docker.com/r/hypriot/rpi-gpio/
http://stackoverflow.com/questions/30059784/docker-access-to-raspberry-pi-gpio-pins

The docker run -v /sys:/sys approach works for enabling https://github.com/JamesBarwell/rpi-gpio.js
Would be great if there was a way to access libraries expecting to use pi-blaster in dockerized apps.

Extra good would be a way to do this with isolated security - either by allocating GPIO pins to a container, or by allowing shared use of pins across containers.

watkinspd, 2016-05-08:
Update: pi-blaster will work with docker if you pass /dev/pi-blaster to the container as a volume.

example: docker run -d -p <port>:<port> -v /dev/pi-blaster:/dev/pi-blaster  <docker image to run>

sarfata, 2016-05-08:
That is even better :) Thanks for reporting success with this!

closed 2016-05-08 by sarfata