sarfata/pi-blaster #63: [please test] Specify gpio list from command line

penfold42, 2016-04-18:
I have an experimental branch which allows you to specify the gpio list from the command line rather than hand editing the code for pi-blaster.c

https://github.com/penfold42/pi-blaster/tree/dynamic-gpio

All feedback welcome !

sarfata, 2016-04-19:
Looks good. Thanks for sharing. 

Let me know when you think it's ready.

sarfata, 2016-04-19:
@penfold42 I have been thinking about this and actually I am not even sure why we limit the number of channels of GPIO. There is a maximum of 32 anyway (that is the size we write in memory to set or read the GPIOs). We could just support all 32 of them and remove a lot of complexity in the program.

Do you see any advantage to keeping a smaller list? Or any advantage to specify the list on the command line? If not, I will probably do the change to remove the channels logic we have right now and just support 32 pins. I will keep the bannedPins idea that you have here, it's a great way to block certain pins.

penfold42, 2016-04-19:
Does it mess with the gpio pin configuration if it's specified but never set from /dev/pi-blaster ?

eg; I need the SPI pins for something else at the same time

sarfata, 2016-04-19:
We would still keep track of which one are used or not. And still support the release command to release them.

It won't mess with anything as long as you do not set them.

penfold42, 2016-04-19:
In some ways I prefer the command line list.

It gives you a defined list of allowed values to stop the clients trashing other gpios accidentally.

penfold42, 2016-04-20:
I think it's good to go now - updated the readme.

#65

git0rhero, 2016-04-20:
If the Pi reboots and automatically starts pi-blaster, does it use previously configured pins or do you have to reconfigure it again from command line?

penfold42, 2016-04-20:
no - the binary is not responsible for remembering the cmd line options.

If you are using the Debian startups scrips, there should be a config file in there

penfold42, 2016-04-20:
/etc/default/pi-blaster looks like the place

fhennig, 2016-04-26:
I cloned it and tested it, works for me as well.
In [line 928](https://github.com/sarfata/pi-blaster/pull/65/files#diff-328b0fc3df32f9fda86573f79733cfb3R928) in `pi-blaster.c` it should be `[-hDgipv]`, the `D` is missing.
I think it should be documented which pins are enabled by default. And the config file (`/etc/default/pi-blaster`) should be mentioned as well, but that mustn't be in this pull request, I'd be happy to document the new features if this pull request is accepted.

I also tested the use of the config file and it worked, though I do not understand the underlying mechanism that makes it work; `DAEMON_OPTS` is used in the config file but never used in `/etc/init.d/pi-blaster`.

penfold42, 2016-04-26:
Thanks, good catch on the usage printf .
I'll amend the usage printf to include an example that just happens to be the default pins.

The DAEMONOPTS is a variable the OS startup scripts look for and automagically appends when it launches the binary.

sarfata, 2016-05-02:
closed by #65. Thanks for helping with tests @fhennig !

closed 2016-05-02 by sarfata