sarfata/pi-blaster #68: Not able to sudo make install on rpi 2 with new jessie

oanylund, 2016-05-19:
I was not able to install when i updated to the new jessie image.

This is what i got when running "sudo make install":
```bash
make[1]: Entering directory '/home/pi/installs/pi-blaster'
 /bin/mkdir -p '/usr/sbin'
  /usr/bin/install -c pi-blaster '/usr/sbin'
 /bin/mkdir -p '/lib/systemd/system'
 /usr/bin/install -c -m 644 pi-blaster.service '/lib/systemd/system'
make  install-data-hook
make[2]: Entering directory '/home/pi/installs/pi-blaster'
systemctl enable pi-blaster
Created symlink from /etc/systemd/system/multi-user.target.wants/pi-blaster.service to /lib/systemd/system/pi-blaster.service.
systemctl start pi-blaster
Job for pi-blaster.service failed. See 'systemctl status pi-blaster.service' and 'journalctl -xn' for details.
Makefile:846: recipe for target 'install-data-hook' failed
make[2]: *** [install-data-hook] Error 1
make[2]: Leaving directory '/home/pi/installs/pi-blaster'
Makefile:771: recipe for target 'install-data-am' failed
make[1]: *** [install-data-am] Error 2
make[1]: Leaving directory '/home/pi/installs/pi-blaster'
Makefile:719: recipe for target 'install-am' failed
make: *** [install-am] Error 2
```

Looked to be an error with systemd. So when i checked the journal with "sudo journalctl -xn" i got this: 
```bash
May 20 01:46:03 raspberrypi systemd[1]: Failed to load environment files: No such file or directory
May 20 01:46:03 raspberrypi systemd[1]: pi-blaster.service failed to run 'start' task: No such file or directory
May 20 01:46:03 raspberrypi systemd[1]: Failed to start Daemon for PWM control of the Raspberry Pi GPIO pins.
```

Seemed to me like the default file was not there, and sure enough, there was no pi-blaster in /etc/default/.

When i copied the file from the repo into where it should have been, sudo make install works fine.
Hope this can help someone, or that the bug gets fixed. 
Cheers for the awesome lib anyways!

wesgann, 2016-05-26:
@oanylund, can you tell me exactly which file you copied over? I'm having the same issue... Thanks in advance!

oanylund, 2016-05-26:
Make a new file at `/etc/default/` named `pi-blaster` with **no extension** (drop the .default from the file in the repostitory).

Copy the contents of [this file](https://github.com/sarfata/pi-blaster/blob/master/debian/pi-blaster.default) into it. 

You should be good to go! :smile:

wesgann, 2016-05-26:
Thanks for the fast reply! I had figured it out by just looking around before I saw that you responded, but this allowed me to confirm my actions were correct. However, it's not working... running echo "18=1" > /dev/pi-blaster doesn't appear to be turning on that pin (it should be turning on the red leds of the light strip) so I don't know if I messed up somewhere else or not. But that's not really related to this issue so I'll keep researching.

oanylund, 2016-05-26:
Np. I hope you get it working!

mhmhartman, 2016-06-08:
@oanylund thank you so much for sharing this! On Jessie lite I came across the exact same issues; your fix solved them and now I've got my servo's up and running :)

conorfennell, 2016-06-08:
@oanylund echo @mhmhartman thanks.

sarfata, 2016-06-12:
Can one of you make a PR to fix this?

sarfata, 2016-08-25:
should be fixed by #76

closed 2016-08-25 by sarfata