sarfata/pi-blaster #69: Light flickers when using less than 1

wesgann, 2016-05-31:
Given
GPIO 18(r), 23(g), 24(b)

If I do `echo 18=0.5 > /dev/pi-blaster` instead of the red led being 50% power, it just flashes. Same behavior for all colors. However, if I issue 18=1, then the light is on full power with no flickering. Where this really causes a problem is getting certain colors such as orange where i need very little green.

sarfata, 2016-05-31:
- Can you give more information about your hardware, software distribution and version of pi-blaster?
- Are you using a screen? Any other peripherals connected?
- Can you give us the output of `pi-blaster` when you start it manually?
- Please try running with the `--pcm` option and let us know if you have problems.

wesgann, 2016-06-01:
Rasp Pi 2 Model B
Distro: Linux rpi-lights 4.4.9-v7+ #884 SMP Fri May 6 17:28:59 BST 2016 armv7l GNU/Linux
Pi-blaster: version 0.1.2
Lights (just in case): http://www.amazon.com/Blue-Wind-Lighting-Waterproof-Controller/dp/B019A3SB30/ref=pd_sim_60_1?ie=UTF8&dpID=61cYEPQ3kkL&dpSrc=sims&preST=_AC_UL160_SR160%2C160_&refRID=0G2D884HZJA876AA4T2P

No screen, I do have a WiFi dongle attached and that's it.

Output:
MBox Board Revision: 0xa21041
DMA Channels Info: 0x7f35, using DMA Channel: 14
Using hardware:                   PWM
Number of channels:                32
PWM frequency:                 100 Hz
PWM steps:                       1000
Maximum period (100  %):      10000us
Minimum period (0.100%):         10us
DMA Base:                  0x3f007000
Initialised, Daemonized, Reading /dev/pi-blaster.

Running `--pcm` seems to work most of time, but there are times when the lights will have less power as they should and then suddenly just start flickering. I have re-worked my code to always run the `--pcm` before setting the pins and this approach seems to be stable. Thanks!

sarfata, 2016-06-12:
> I have re-worked my code to always run the --pcm before setting the pins and this approach seems to be stable.

You should only start pi-blaster once and never need to re-run it later. Do you kill it or stop it?

jorupp, 2016-06-14:
If you're finding that it flickers with non-0/1 values and that restarting the pi-blaster daemon resolves it (temporarily), it sounds like you're having the same issue I did (#67) - it seemed something else I was running messed with the PWM config.  Switching to PCM (as you've tried here) got it working for me.