sarfata/pi-blaster #71: systemd problems

nhooyr, 2016-06-11:
why is `/etc/default/pi-blaster` not created automatically before enabling/starting the service in the makefile? A simple `touch /etc/default/pi-blaster` is all that is required.

Also on arch linux, you cannot run makepkg as root and so when `systemd enable pi-blaster.service` is run [here](https://github.com/sarfata/pi-blaster/blob/d2869cee24903998a6e8baf3387b6238064e874b/Makefile.am#L17), it fails. Perhaps change this to `sudo systemd enable pi-blaster.service`?

sarfata, 2016-06-12:
Thanks for the comments. I am not an Arch user so I cannot test this. Can you make a pull-request?

doudz, 2016-07-28:
sudo touch /etc/default/pi-blaster did the trick on my fresh installed jessie on rpi 2

nhooyr, 2019-11-21:
Going to close as it has been a long time and I'm not sure if this is relevant anymore.

closed 2019-11-21 by nhooyr