sarfata/pi-blaster #72: Job for pi-blaster failed

Bubba, 2016-06-21:
When running `sudo make install` I get this error:

```
 /bin/mkdir -p '/usr/sbin'
  /usr/bin/install -c pi-blaster '/usr/sbin'
 /bin/mkdir -p '/lib/systemd/system'
 /usr/bin/install -c -m 644 pi-blaster.service '/lib/systemd/system'
make  install-data-hook
make[2]: Entering directory '/home/pi/Downloads/pi-blaster'
systemctl enable pi-blaster
systemctl start pi-blaster
Job for pi-blaster.service failed. See 'systemctl status pi-blaster.service' and 'journalctl -xn' for details.
Makefile:846: recipe for target 'install-data-hook' failed
make[2]: *** [install-data-hook] Error 1
make[2]: Leaving directory '/home/pi/Downloads/pi-blaster'
Makefile:771: recipe for target 'install-data-am' failed
make[1]: *** [install-data-am] Error 2
make[1]: Leaving directory '/home/pi/Downloads/pi-blaster'
Makefile:719: recipe for target 'install-am' failed
```

Running `systemctl status pi-blaster.service` gives me

```
pi-blaster.service - Daemon for PWM control of the Raspberry Pi GPIO pins
   Loaded: loaded (/lib/systemd/system/pi-blaster.service; enabled)
   Active: failed (Result: resources)
```

and `journalctl -xn` gives me `No journal files were found.`

Running it as `sudo ./pi-blaster` works fine:
```
MBox Board Revision: 0xa22082
DMA Channels Info: 0x7f35, using DMA Channel: 14
Using hardware:                   PWM
Number of channels:                32
PWM frequency:                 100 Hz
PWM steps:                       1000
Maximum period (100  %):      10000us
Minimum period (0.100%):         10us
DMA Base:                  0x3f007000
Initialised, Daemonized, Reading /dev/pi-blaster.
```
But I still get the same result from `systemctl status pi-blaster.service`

doudz, 2016-07-28:
sudo touch /etc/default/pi-blaster did the trick on my fresh installed jessie on rpi 2

deadsy, 2016-08-08:
Same issue on jessie/rpi3.
It wants to see /etc/default/pi-blaster for a good startup.

sarfata, 2016-08-25:
should be fixed by #76

closed 2016-08-25 by sarfata