sarfata/pi-blaster #73: strange behavior runnig headless

yanoschik, 2016-07-07:
I've run into a strange issue...
Using pi-blaster to control two servos in a camera pan-tilt setup ([here](https://github.com/yanoschik/PiPan_Cam)). RPi 2 running Raspbian Jessie, Linux raspberrypi 4.4.13-v7+ #894

Everything works fine, except when starting up headless - only in that case, the servos go crazy = a full power spasm. I believe it's the PWM firing up 100%. There is a script that sets a default servo position at startup, and it runs fine as I can see the camera going into position. But in a moment, the servos go nuts and I have to pull the plug as not to fry them :(
When I start up with the monitor connected to HDMI - no problem.
Any ideas why this happens and where/what to investigate further?
Thanks

penfold42, 2016-07-07:
Yes ... And kinda.

We've got the same problem with the rpi_281x library.

My best guessing the timers are set up differently by the firmware depending on the presence or absence of an HDMI connector.

Short term workaround I've heard of but not tried is to plug in a HDMI to micro HDMI adaptor - that's enough to fool it

yanoschik, 2016-07-07:
I've tried to plug in an active HDMI->VGA adapter ...bud it did not help :(
(well, but I did not power it up, got it [modified for external power](http://imgur.com/a/sLogs#0) because of [powering problems](http://elinux.org/R-Pi_Troubleshooting#HDMI_-.3E_VGA_adapters), so I'll try that again later)

sarfata, 2016-07-07:
We are not sure how to explain that but could you try with `--pcm` when launching pi-blaster? This will use a different timing source that seems to work better.

Also can you confirm what version of RaspberryPi you are using?

yanoschik, 2016-07-07:
Could you explain a bit more (o refer to) what is and how to use `--pcm`? I'm not that good with this stuff yet ;)  Only managed to google that it might be something regarding PCM/I2S clock for audio...

And it's Raspberry Pi 2 Model B ...as I already wrote in the first post.
pi-blaster is launching with reboot as I've used `sudo make install`

yanoschik, 2016-07-08:
Just a note: it works OK with the **powered** HDMI->VGA adapter

yanoschik, 2016-09-29:
Well, the `--pcm` [option](https://github.com/sarfata/pi-blaster/blob/master/README.md#options) seems to work fine...
It just took a while for me to figure out how to use it :/
I've added a line with `DAEMON_OPTS='--pcm'` to `/etc/default/pi-blaster` _(as this file was empty in my case, and I did not know first where and how to pass that argument)_.