sarfata/pi-blaster #74: Cannot use pin after releasing it

ChrisTrenkamp, 2016-07-20:
I'm running a minimal Gentoo installation with Busybox on a RPI B+.  When starting the daemon for the first time, everything's fine.  After releasing a pin and try turning it on again, it doesn't work.  The pwm frequency operates very funkily, and after releasing the pin again, it doesn't work at all.

I thought this was a feature at first, but after using this fork: [https://github.com/hybridgroup/pi-blaster](https://github.com/hybridgroup/pi-blaster), I don't see the same behaviour.  The pin starts and stops with no problem.

sasanrose, 2017-02-28:
Same problem here. I am using minibian on RPI B+

sasanrose, 2017-02-28:
I tried to do a git bisect and seems that the problem was introduced in e0cb6de39798e6ac2f6dfacffc4b468dffda4ffb

I think when you run `pi-blaster -D` (Which means no daemonize) then everything works finde

ollyhayes, 2017-05-04:
Yeah this is happening to me too, not every time I release, but most times. Sometimes it just one pin (but always the same one), but most of the time it's both. I have to stop and start the daemon to get it work again.

For me running pi-blaster -D without the daemon running made no difference. Did you ever get to the bottom of it?

I'm on a raspberry pi 3, running raspbian, and I'm driving two externally powered servo motors.

DaBlitzStein, 2017-10-11:
Here Pi 3 with raspbian and the same with or without -D . When my gobot program stops, I have to restart pi-blaster service to use these pins.

dankozlowski, 2018-05-02:
The issue with setting it to `0` is that the line is still open, and you can get minor jitter.

bezineb5, 2018-08-27:
This issue should be resolved now, with the PR #101

sarfata, 2018-08-31:
Thank you @bezineb5 !

closed 2018-08-31 by sarfata