sarfata/pi-blaster #75: Noise on sound jack when starting pi-blaster

qbalsdon, 2016-08-22:
So I cannot run make install, I get:
Makefile:719: recipe for target 'install-am' failed

But that is not my issue - because I can still run
sudo ./pi-blaster

However as soon as I do there is a high pitched beep that comes through my audio jack. If I start the pi-blaster from /home/pi/.config/lxsession/LXDE-pi/autostart - the noise sounds more like a signal noise.

fenwick67, 2016-09-24:
Isn't this expected?

> This daemon uses the hardware PWM generator of the raspberry pi to get precise timings. This might interfere with your sound card output

yanoschik, 2016-09-30:
...I've used the `--pcm` [option ](https://github.com/sarfata/pi-blaster/blob/master/README.md#options) too, to solve a different problem - [see here](https://github.com/sarfata/pi-blaster/issues/73#issuecomment-250589367)

penfold42, 2016-09-30:
Maybe this should be the default ?

yanoschik, 2016-10-03:
Actually that did not help with audio in my case, only solved the other problem...
I get a strong pulsating noise/interference on the jack. And that makes analog audio virtually useless :/