sarfata/pi-blaster #79: apt-get:  unable to locate package pi-blaster?

lettcco, 2016-12-08:
Hello,  I am a pi / Linux newbie.  So far all the instruction I came across on how to install new apps are thru 
sudo apt-get install xxxxxxxx
So I across this GitHub while trying to figure out how to drive servo...I am not sure what I did wrong but I can't seems to install the package.  Am I doing something wrong?  Thanks in advance.

JamesDaniel, 2017-01-05:
I'm having the same problem.

```
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package pi-blaster
```
I tried :
```
sudo apt-get update
sudo apt-get dist-upgrade
sudo apt-get install pi-blaster
```
but I get the same output.
System info:
```
Linux raspberrypi 4.4.34+ #930 Wed Nov 23 15:12:30 GMT 2016 armv6l GNU/Linux
```
and
```
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

DANDWYERIRL, 2017-01-09:
just tried installing and i'm experiencing the same issue as above.  On a freshly imaged pi. Think this may be an issue

mpowers3, 2017-01-18:
My error message is the same as @JamesDaniel. Has anyone found a way to work around this, or install pi-blaster?

sarfata, 2017-01-20:
All, thanks for reporting this. Unfortunately this package was submitted to debian by 
@hosiet and I am not sure why it's not available anymore.

You can still install pi-blaster manually. Just follow the instructions in the readme:

Build the package yourself:

    sudo apt-get install debhelper dh-autoreconf dh-systemd dpkg-dev init-system-helpers autoconf
    dpkg-buildpackage -us -uc -i 

Install it:

    sudo dpkg -i ../pi-blaster*.deb

hosiet, 2017-01-20:
Sorry but I think I have never ever reported this package into Debian... Why would you have such an impression? :-p

sarfata, 2017-01-20:
Wow. Just re-read the history of our pull-request and yes I am not sure why I believed that... Sorry.

I will remove the `apt-get install` from the README.

sarfata, 2017-01-20:
README fixed. Sorry for leading people in the wrong direction. The manual install instructions are the proper way to go. I am not interested in submitting and maintaining a package with Raspbian. 

If someone wants to do it, please just let me know and I will re-update the readme.

closed 2017-01-20 by sarfata