sarfata/pi-blaster #83: cannot open 'debian/changelog' for reading

ash73, 2017-01-29:
Hi, trying to install pi-blaster on a Pi2 with the following commands:

sudo apt-get install debhelper dh-autoreconf dh-systemd dpkg-dev init-system-helpers autoconf
dpkg-buildpackage -us -uc -i && sudo dpkg -i ../pi-blaster*.deb

The first line completes ok, the second gives the following error message:

tail: cannot open ‘debian/changelog’ for reading: No such file or directory
dpkg-buildpackage: error: tail of debian/changelog gave error exit status 1

Can anyone help? TIA

hosiet, 2017-01-29:
Before you run `dpkg-buildpackage`, please change your working directory into "pi-blaster" source dir. (something like `cd pi-blaster/`)

ash73, 2017-01-29:
It would help if I actually downloaded the source files! Have now found the clone command; sorry, new to this.

closed 2017-01-29 by ash73