sarfata/pi-blaster #86: tutorials/examples?

Ch3fk0ch88, 2017-04-18:
Hi,
are there some tutorial/example script for Pi-Blaster.(dim an LED or something like that?).

penfold42, 2017-04-18:
Did you download the source code and cd into the pi-blaster directory ?

Ch3fk0ch88, 2017-04-18:
Ye, after a bit of search i found the way to install it.

But are the python or java exampels`?

Ch3fk0ch88, 2017-04-19:
Did no one have an idear where i can find an example?
I find nothing with pi-blaster.

git0rhero, 2017-04-19:
What you need is an example of how to write to a file path in those languages. Then you just need to write to /dev/pi-blaster . Same as you would with the shell command 'echo' in the example in the readme.

Ch3fk0ch88, 2017-04-19:
Ok i will search for it (I wanted to use python)

Ch3fk0ch88, 2017-04-19:
Ok, to write something in a fifo file i need the name of it xxxx.fifo.
But what is the name?
Or create "echo "17=1" > /dev/pi-blaster" a file?

I dont check it.

git0rhero, 2017-04-19:
The name is /dev/pi-blaster. It is created when you run 'sudo ./pi-blaster'
Is it that what you mean or am I on a wood way? ;)

Ch3fk0ch88, 2017-04-19:
But i can't open a file "/dev/pi-blaster" cause it is no file.
A File would be pi-blaster.fifo

Ch3fk0ch88, 2017-04-22:
No one?

RunnicFusion, 2018-07-08:
What todo after installing autoconf? What is the next step to get pi blaster installed? Git cloned the repo into dev/pi-blaster

gherlein, 2018-07-08:
@RunnicFusion this is probably not the forum to ask basic howto compile questions... did you follow the instructions at https://github.com/sarfata/pi-blaster#build-and-install-directly-from-source ???

gherlein, 2018-07-08:
@Ch3fk0ch88  are you sure you have installed it correctly?  do a 'sudo ps aux|grep blaster' and you should see it running.  If it's running, can you do the steps at https://github.com/sarfata/pi-blaster#how-to-use ???  There are detailed instructions.  If you follow them and post your outputs and ask clear questions we can likely help you.  You really have not provided much about what you see or don't see.  Consider reading:  http://www.catb.org/esr/faqs/smart-questions.html

sarfata, 2018-07-09:
@gherlein great ESR link, thanks for sharing!

closed 2018-07-09 by sarfata