sarfata/pi-blaster #88: GPIO not enabled for pi-blaster

SephReed, 2017-08-07:
This is the full list of non-banned GPIOS `[17,18,16,19,13,20,12,21,22,5,23,4,24,25,26,27]`
If you run `$  pi-blaster --gpio 17,18,16,19,13,20,12,21,22,5,23,4,24,25,26,27` it will return no errors

But if you run a script to set each of these GPIO to PW 0.5, the following will throw an Error: `GPIO #num is not enabled for pi-blaster`  --> `[16,19,13,20,12,5,26]`

Can these GPIOs be enabled for pi-blaster?

GPIOS:
> all :: [17,18,16,19,13,20,12,21,6,22,5,23,4,24,25,26,27]
> banned :: [6,]
> non banned :: [17,18,16,19,13,20,12,21,22,5,23,4,24,25,26,27]
> non useable :: [16,19,13,20,12,5,26]
> useable :: [17,18,21,22,23,24,4,25,27]

SephReed, 2017-08-07:
Figured it out: `sudo` must be used for some of the GPIOs, but not all.

To solve, simply add `sudo`

`$ sudo pi-blaster --gpio 17,18,16,19,13,20,12,21,22,5,23,4,24,25,26,27`

sarfata, 2017-08-08:
Thanks for sharing the solution!

closed 2017-08-08 by sarfata