sarfata/pi-blaster #89: Error at raspberry pi 3

Temeteron, 2017-08-17:
I am using this library via a [tutorial](https://github.com/craig-mulligan/resin-servo-node), which is working for everyone but me. So i thought that the problem is probably at the pi-blaster library.

After running:
`./pi-blaster 22 && echo "22=1" > /dev/pi-blaster`
where pi-blaster is the executable of the pi-blaster.c,
i got the following output:
```
Using hardware: PWM
Number of channels: 1
PWM frequency: 100 Hz
PWM steps: 1000
Maximum period (100 %): 10000us
Minimum period (0.100%): 10us
pi-blaster: Page 0 not present (pfn 0xa100000000016864)
```

I am not understanding what is the meaning of the last line. ???

By trying the command:
`echo "22=1" > /dev/pi-blaster`
i got no output and the servo motor didn't work.

Thanks in advance

Temeteron, 2017-08-18:
I found the problem. The tutorial i was using probably had an older version of this library.

Nevertheless, please comment about the meaning of the last line of the output:
`pi-blaster: Page 0 not present (pfn 0xa100000000016864)`

mvelebit, 2018-02-13:
Could you post the fix which you utilized? Thanks in advance

Temeteron, 2018-02-14:
This worked for me:

```
echo "0=0.1" > /dev/pi-blaster
```
A full running example on an aquarium can be found [here](https://github.com/Temeteron/aquarium_raspberry), which contains an implementation that combines some of the tutorials of [resin](https://docs.resin.io/getting-started/raspberrypi3/nodejs/). More specifically a nodeJs server with a frontend and the pi-blaster module to control a servo motor through a web page and eventually feed your fishes.

polymathuk, 2021-09-07:
I am having the exact same problem. The command outputs:
```
Using hardware:                   PWM
Number of channels:                 1
PWM frequency:                 100 Hz
PWM steps:                       1000
Maximum period (100  %):      10000us
Minimum period (0.100%):         10us
pi-blaster: Page 0 not present (pfn 0xa100000000015c1e)
```
The file pi-blaster.c contains a function make_pagemap(void) which I assume generates the error line, but I have no idea what it means. I'm not getting a process running in top for pi-blaster and assigning values like:
`echo "1=0.2" > /dev/pi-blaster`
has no effect on the actual pins.
Using a 1.12 Pi from 2011