sarfata/pi-blaster #92: Is it possible to do `sudo make install` with --pcm activated?

AnduriI, 2018-01-21:
Hey there,

I had some problems with LEDs working fine after boot, but flickering after some time. As supposed by @jorupp in #69 this can be solved by using `--pcm` option. I tested it and it works really well.
Is there a way to autostart the daemon with --pcm on every boot? At the moment I have it running with `sudo make install` and I do not need to care about. I know I could use crontab to start it manually @reboot, but I hope there is a more elegant way to do.

Regards,
Anduril

jorupp, 2018-01-21:
Someone who knows a bit more about how this is configured on a Pi can probably help more, but I recall there being a pi-blaster.service file in /etc/systemd/system or somewhere like that that had a line `ExecStart=` that indicated the executable to run - you can just add additional parameters to that line, such as the `--pcm`.

Good luck.

AnduriI, 2018-01-21:
Thanks for your hint, I found the `pi-blaster.service` in `/etc/systemd/system/multi-user.target.wants/`. As you thought it contains this line, but I don't know what `$DAEMON_OPTS`is and what it does. Should I add `--pcm` before or after that?

```
[Unit]
Description=Daemon for PWM control of the Raspberry Pi GPIO pins

[Service]
EnvironmentFile=-/etc/default/pi-blaster
User=root
ExecStart=/usr/sbin/pi-blaster $DAEMON_OPTS
ExecReload=/bin/kill -s HUP $MAINPID
Type=forking

[Install]
WantedBy=multi-user.target
```

jorupp, 2018-01-21:
Oh yeah, forgot about DAEMON_OPTS - maybe `/etc/default/pi-blaster` or something like that defines that?  I'm *far* from a systemd expert :(

sarfata, 2018-01-26:
Yes that is the solution :)

closed 2018-01-26 by sarfata