sarfata/pi-blaster #95: no longer seems to install on raspberry pi 3

carl-ranson, 2018-03-31:
having trouble installing this on a raspberry pi 3. 

I've been trying the dpkg instructions in the readme. The first command appears to work fine, but this one: 
pi@raspberrypi:~/go/src $ dpkg-buildpackage -us -uc -i && sudo dpkg -i ../pi-blaster*.deb

produces this error message. 
tail: cannot open 'debian/changelog' for reading: No such file or directory

googling this error suggests that the debian folder was changed to debianpkg some time late in 2017. 
see: https://github.com/FRRouting/frr/issues/1485

is the install script out of date due to this change?

dankozlowski, 2018-04-19:
I've just installed the package manually, although I'm having trouble getting anything to work.

dankozlowski, 2018-04-20:
I've confirmed that installing the application manually works on the Pi3.

carl-ranson, 2018-04-22:
Great, thanks Dan, 
can you confirm which install method you used? which version of the os?

dankozlowski, 2018-04-23:
I followed the manual installation steps (compiling from source, NOT creating a *.deb) on the latest Raspbian.

To do this, clone the repository down (or download the zip and unzip it somewhere on your Pi), and follow the instructions under https://github.com/sarfata/pi-blaster#build-and-install-directly-from-source.

carl-ranson, 2018-04-24:
ok Dan. I got it to work using that method. Still doesn't address that fact that the other technique is maybe out of date, but i'm good for now so moving on. 
thanks again. 
CR

dankozlowski, 2018-05-02:
Can we close this issue now?

carl-ranson, 2018-05-03:
Well I would suggest an update to the documentation noting that the dpkg instructions may not work on a pi 3b and to try building from source if that's the case, just in case anyone comes after me with the same problem.  

other than that, I've got no problem with closing the issue. 

thanks for your help