sarfata/pi-blaster #97: pi 3B+

kennylouth, 2018-04-30:
pi-blaster not working on new pi 3B+.
All software is same using latest raspbian stretch.
Simply transferred SD card from pi 3B to pi 3B+.
On new 3 B+ 'systemctl status pi-blaster' shows pi-blaster as running normally.
Hopefully this is a simple fix

kennylouth, 2018-05-03:
Yes apparently the gpio pins are not being set to outputs.
Systemd  systemctl status pi-blaster.service is running OK.
Both pi-blaster version 1.1 and 1.2 do not work on the new pi 3B+
A update to wiringpi was necessary for wiringpi to work on the pi 3B+.
Hope this helps.

kennylouth, 2018-05-13:
> Are you able to provide any more diagnostic information?

Yes apparently the gpio pins are not being set to outputs.
Systemd systemctl status pi-blaster.service is running OK.
Both pi-blaster version 1.1 and 1.2 do not work on the new pi 3B+
A update to wiringpi was necessary for wiringpi to work on the pi 3B+.
Hope this helps.

gherlein, 2018-05-13:
This does not really make sense to me - what does WiringPi have to do with
pi-blaster?

--------------------------------------------------------------------------------------------------------------
Greg Herlein   |   415-368-7546   |   LinkedIn
<http://www.linkedin.com/in/gherlein>

kennylouth, 2018-05-13:
Below is the result running pi-blaster in bash.
Hope this helps.

MBox Board Revision: 0xa020d3
DMA Channels Info: 0x7f35, using DMA Channel: 14
Using hardware:                   PWM
Number of channels:                32
PWM frequency:                  50 Hz
PWM steps:                       1000
Maximum period (100  %):      20000us
Minimum period (0.100%):         20us
DMA Base:                  0x20007000
pi-blaster: Failed to open /dev/mem: Permission denied
pi-blaster stopped.

ssokol, 2018-05-14:
I can confirm the issue. I've not had any issues with /dev/mem. I simply don't get any output on the assigned pins on a 3 B+. The same code has been working perfectly on the Pi 3 B units we've been deploying for the past year.

penfold42, 2018-05-15:
Oops - don’t know why my merged PR closed this.

Please these with the latest code )as of 5mins ago)

sarfata, 2018-05-15:
> Oops - don’t know why my merged PR closed this.

It's because your PR had the words `Fix #97` in the description. Github is _that_ smart!

See https://help.github.com/articles/closing-issues-using-keywords/

kennylouth, 2018-05-17:
After download from git today
both ...
./autogen.sh
./configure
compile fine
but make fails see below
make  all-am
make[1]: Entering directory '/home/pi/pi-blaster-master'
gcc -DHAVE_CONFIG_H -I.     -Wall -pedantic -std=c99 -MT pi-blaster.o -MD -MP -MF .deps/pi-blaster.Tpo -c -o pi-blaster.o pi-blaster.c
In file included from /usr/include/stdio.h:27:0,
                 from pi-blaster.c:31:
/usr/include/features.h:148:3: warning: #warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE" [-Wcpp]
 # warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE"
   ^~~~~~~
mv -f .deps/pi-blaster.Tpo .deps/pi-blaster.Po
gcc  -Wall -pedantic -std=c99   -o pi-blaster pi-blaster.o mailbox.o  
make[1]: *** No rule to make target 'pi-blaster.service', needed by 'all-am'.  Stop.
make[1]: Leaving directory '/home/pi/pi-blaster-master'
Makefile:317: recipe for target 'all' failed
make: *** [all] Error 2

sudo make install therefore also fails

but sudo ./pi-blaster works just fine.

PR #98 (2018-05-15): Add Pi 3B+ support
Fix #97
closed 2018-05-17 by kennylouth