sarfata/pi-blaster #105: PWM pin usable

BILLyTheLiTTle, 2018-09-21:
I don't know where to ask  question, so I opened this issue.
I read that the blaster uses the hardware pwm. I also read that there will be an issue with the sound and that I cannot use the pwm pin as an input.
But, does the blaster driver affects the usage of the pwm pins when I want to send pwm through the pwm pins?
To explain myself better, can I control 4 pwm devices (motors) and 8 blaster devices (servos) simultaneously without using any SoftPwm?

sarfata, 2018-11-19:
Actually, I am not sure. Can you give it a try and let us know?

BILLyTheLiTTle, 2018-11-19:
I am planning to do so. I am going to use it at my current project but this will be one of the lasts tests. 
I will update this thread no matter the result.