sarfata/pi-blaster #106: Pi-Blaster dont Work?

Ch3fk0ch88, 2018-09-28:
Hi,
I dont know why but i cant controll the Fan.
I installed it, than go in as Root cause as normal Pi user with Sudo i all time get a permission Error.

So i go to the pi-Blaster folder than ./pi-blaster ant this gives me:
`
Using hardware:                   PWM
Number of channels:                 8
PWM frequency:                 100 Hz
PWM steps:                       1000
Maximum period (100  %):      10000us
Minimum period (0.100%):         10us
pi-blaster: Page 0 not present (pfn 0xa100000000027053)
`
But when i try to control a the Bin BCM 12 nothing happens
`root@raspberrypi:/home/pi/pi-blaster# echo "12=0" > /dev/pi-blaster
`
I hope someone can help me.

Thanks

Ch3fk0ch88, 2018-09-29:
Nobody who can help?

penfold42, 2018-09-29:
Page 0 not present (pfn 0xa100000000027053)

That is probably why it doesn’t work.

No idea what it means tho !

Did you build from source ?

sarfata, 2018-10-02:
@Ch3fk0ch88 can you tell us which raspberry pi you are using? What is the exact hardware?

Also, what OS and version? And as penfold42 asked, how did you install pi-blaster?

Ch3fk0ch88, 2018-10-02:
Hi,
its a RPI3b+ with an sandisk Ultra 16GB A1.
Operating system is Rasbian (fresh downloaded), updated and upgraded.
I Installed it on this way:
`pi@raspberrypi ~ $ git clone https://github.com/mwilliams03/pi-blaster.git
pi@raspberrypi ~ $ cd pi-blaster
pi@raspberrypi ~ $ make pi-blaster`

Greetings

sarfata, 2018-10-02:
Thanks. This might be a problem on the 3b+. I don't have one to test with. @penfold42 do you have one?

penfold42, 2018-10-03:
I do - I’ll see what I can try in the next few days

Ch3fk0ch88, 2018-10-04:
Hm seems i made an mistake.
Did everythign new and it works.