sarfata/pi-blaster #112: Conflict with Pi-Somfy

suhajda3, 2020-12-05:
I use [Pi-Somfy](https://github.com/Nickduino/Pi-Somfy) to control my shutters and [homebridge-gpio-rgb-ledstrip](https://github.com/manfredipist/homebridge-gpio-rgb-ledstrip) to control my LED strip. homebridge-gpio-rgb-ledstrip is using pi-blaster that is why I created the issue here.

It looks like that Pi-Somfy and pi-blaster has some kind of conflict. My LED strip uses GPIO pins 17, 22, 24 and I have setup pi-blaster to control only these:
DAEMON_OPTS="--gpio 17,22,24"

Pi-Somfy uses GPIO 4.

If I used my LED's before and want to control my shutters I have to restart Pi-Somfy. After restart the shutters work but the LED strip no longer can be controlled via homebrdige. If I restart pi-blaster the LED strip can be controlled again but the shutters won't work.

I believe there is some kind of conflict between the two.

sarfata, 2020-12-06:
_thanks for pointing me to pi-somfy - cool project!_

I think the conflict here is that pi-somfy uses pygpio which uses [pigpio](https://github.com/joan2937/pigpio). I had not seen this project before but it looks like a project very similar to pi-blaster (with more features and better doc ;P).

I do not think this is a GPIO problem but probably more a DMA channel problem. Are you on a raspberry pi 4? Pi-blaster uses channel 7 by default on the 4 and so does pigpio (according to their doc). You will want to change the channel for one of the program and see if it works afterwards.

With pigpio, they say:
> You can set the channels used by the pigpio daemon by invoking it with the -d and -e options, e.g. sudo pigpiod -d 5 -e 8 to specify primary 5, secondary 8.

With pi-blaster you will need to edit the channel in the source code and recompile: 
https://github.com/sarfata/pi-blaster/blob/master/pi-blaster.c#L130

Let me know if you are able to get it to work!

suhajda3, 2020-12-06:
Thanks I also think DMA channels could be a problem but I still couldn't fix the issue.

I am using a Raspberry Pi Model 4B.

I don't know how Pi-Somfy invokes pigpiod so I have enabled the pigpiod daemon (sudo systemctl enable pigpiod) and edited it's service (/lib/systemd/system/pigpiod.service) to use the DMA channels you described:
`ExecStart=/usr/bin/pigpiod -l -d 5 -e 8`

Is there a way to view which DMA channels are in use?

sarfata, 2020-12-06:
That should be good. 

If you turn off pi-blaster, does pigpiod works on this channel? Unfortunately, I never found a good way to figure out which DMA channels are in use. I think they have the same issue. So we just pick something that seems free.

They say: 
> At the moment pigpio on the Pi4B is experimental. I am not sure if the DMA channels being used are safe. The Pi4B defaults are primary channel 7, secondary channel 6. If these channels do not work you will have to experiment. You can set the channels used by the pigpio daemon by invoking it with the -d and -e options, e.g. sudo pigpiod -d 5 -e 8 to specify primary 5, secondary 8.

Can you try with 6 and 5? The first step of this experiment would be to find another DMA channel that works without pi-blaster and then try with pi-blaster.

suhajda3, 2020-12-08:
I have run some tests today. This is what I found:

First I changed the DMA to channel 5 in pi-blaster. Same result as before. pi-blaster and Pi-Somfy conflicted each other.

Then I changed it channel 6. I was a bit skeptical about this one because I read at a [forum post](https://www.raspberrypi.org/forums/viewtopic.php?t=86339):

> Avoid channels 0, 1, 2, 3, 6, 7. The GPU uses 1, 3, 6, 7. The frame buffer uses 0 and the SD card uses 2.

The same thing happened. pi-blaster and Pi-Somfy conflicted each other.

Finally I changed it to channel 8. In this case Pi-Somfy never worked.

I am now not sure that this is a DMA problem. Under channel 8 there is clearly a DMA issue and pi-blaster blocks pigpio.

sarfata, 2020-12-08:
I am afraid the best way to resolve this would be to use pigpiod to control your rgb strip. You can try forking `homebridge-gpio-rgb-ledstrip` to do this. It should not be too hard, both programs work in a very similar way.