sarfata/pi-blaster #116: pi blaster cannot be installed

desinox, 2021-08-21:
Hello, 
i have been trying to install Pilaster on a Raspberry Pi 4b (running raspbian 10 with Homebridge) for quite some time now. 
When I install it via the deb package I don't get an error message, but pi-blaster doesn't work afterwards. When I try to install via source, I always get the following error when running pi-blaster:

MBox Board Revision: 0xc03112
DMA Channels Info: 0x37f5, using DMA Channel: 7
Using hardware:                   PWM
Number of channels:                32
PWM frequency:                 100 Hz
PWM steps:                       1000
Maximum period (100  %):      10000us
Minimum period (0.100%):         10us
DMA Base:                  0xfe007000
pi-blaster: Failed to open /dev/mem: Operation not permitted
pi-blaster stopped.

I hope someone can help me, 
Simon

AdrianAguilarT, 2021-08-23:
I’m getting same error agree install on RPi 3 B.
The only way to get it working is using “sudo pi-blaster”. But i need to do this every time Raspberry restarts.

hugomon, 2022-01-08:
Same problem with RPi 3 B and compiled for Raspios Bullseye arm64

swdee, 2022-04-14:
Note that you need to run pi-blaster as root/superuser for it to work as it requires access to /dev/mem to enable PWM.