sarfata/pi-blaster #117: Is there anywhere C++ example for using pi-blaster?

CrackAndDie, 2021-11-05:
I were using Python before and  had the same problem - I couldn't find Python example anywhere. I tried a lot, because simple writing to file wasn't succesful. So now I have an example for python, that could be added to README file:
```python
    @staticmethod
    def echo_to_file(st: str):
        original_stdout = sys.stdout
        with open('/dev/pi-blaster', 'w') as f:
            sys.stdout = f  # Change the standard output to the file we created.
            print(st)
            sys.stdout = original_stdout  # Reset the standard output to its original value
```
Can someone share C++ example with me if You have it?

CrackAndDie, 2021-11-05:
Found something helpful here: https://ozzmaker.com/software-pwm-on-a-raspberry-pi/. Sorry for opening the issue.

closed 2021-11-05 by CrackAndDie

sarfata, 2021-11-05:
Hi @CrackAndDie 

If you have time to make a PR to add an example in python and another one in C++ I would be happy to merge it. Or maybe just links to the examples.

thanks!
thomas

CrackAndDie, 2021-11-06:
I'm not so good at Git and GitHub so I dunno how to make normal PR. So I will just paste Python example here: https://pastebin.com/69FMBjEQ and C++ example here: https://pastebin.com/HDBd4ZKT. It would be greatful if you add it into README file so other users of pi-blaster will know how to use it in Python and C++ :)

sarfata, 2021-11-08:
@CrackAndDie Thanks. Your examples just show how to write in a file. I think this is too generic to be really valuable in the README.

CrackAndDie, 2021-11-11:
@sarfata A little strange, because you have the same example for Java in your README file, well, ok. As you wish :)

sarfata, 2021-11-12:
Very true! My apologies, it's been a long time since I wrote that README.

I have added your two examples and added your name to the contributors list.

thank you.

CrackAndDie, 2021-11-12:
thank you too :)