sarfata/pi-blaster #119: pi-blaster stops without error

whiskthecat, 2023-01-07:
root@DietPi:~/pi-blaster# pi-blaster
MBox Board Revision: 0xa020d3
DMA Channels Info: 0x7ff5, using DMA Channel: 14
Using hardware:                   PWM
Number of channels:                32
PWM frequency:                 100 Hz
PWM steps:                       1000
Maximum period (100  %):      10000us
Minimum period (0.100%):         10us
DMA Base:                  0x3f007000
pi-blaster stopped.

-------------------------------------------------------------------------------------------------------

root@DietPi:~/pi-blaster# systemctl status pi-blaster
● pi-blaster.boot.service - LSB: Daemon for PWM control of the Raspberry Pi GPIO pins
     Loaded: loaded (/etc/init.d/pi-blaster.boot.sh; generated)
     Active: active (exited) since Fri 2023-01-06 21:37:32 CST; 21min ago
       Docs: man:systemd-sysv-generator(8)
    Process: 273 ExecStart=/etc/init.d/pi-blaster.boot.sh start (code=exited, status=0/SUCCESS)
        CPU: 96ms

Jan 06 21:37:32 DietPi pi-blaster.boot.sh[284]: DMA Channels Info: 0x7ff5, using DMA Channel: 14
Jan 06 21:37:32 DietPi pi-blaster.boot.sh[284]: Using hardware:                   PWM
Jan 06 21:37:32 DietPi pi-blaster.boot.sh[284]: Number of channels:                32
Jan 06 21:37:32 DietPi pi-blaster.boot.sh[284]: PWM frequency:                 100 Hz
Jan 06 21:37:32 DietPi pi-blaster.boot.sh[284]: PWM steps:                       1000
Jan 06 21:37:32 DietPi pi-blaster.boot.sh[284]: Maximum period (100  %):      10000us
Jan 06 21:37:32 DietPi pi-blaster.boot.sh[284]: Minimum period (0.100%):         10us
Jan 06 21:37:32 DietPi pi-blaster.boot.sh[284]: DMA Base:                  0x3f007000
Jan 06 21:37:32 DietPi pi-blaster.boot.sh[284]: pi-blaster stopped.
Jan 06 21:37:32 DietPi systemd[1]: Started LSB: Daemon for PWM control of the Raspberry Pi GPIO pins.

-------------------------------------------------------------------------------------------------------------

root@DietPi:~/pi-blaster# make
make  all-am
make[1]: Entering directory '/root/pi-blaster'
gcc -DHAVE_CONFIG_H -I.     -Wall -pedantic -std=c99 -MT pi-blaster.o -MD -MP -MF .deps/pi-blaster.Tpo -c -o pi-blaster.o pi-blaster.c
In file included from /usr/include/aarch64-linux-gnu/bits/libc-header-start.h:33,
                 from /usr/include/stdio.h:27,
                 from pi-blaster.c:43:
/usr/include/features.h:187:3: warning: #warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE" [-Wcpp]
  187 | # warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE"
      |   ^~~~~~~
mv -f .deps/pi-blaster.Tpo .deps/pi-blaster.Po
gcc -DHAVE_CONFIG_H -I.     -Wall -pedantic -std=c99 -MT mailbox.o -MD -MP -MF .deps/mailbox.Tpo -c -o mailbox.o mailbox.c
mv -f .deps/mailbox.Tpo .deps/mailbox.Po
gcc  -Wall -pedantic -std=c99   -o pi-blaster pi-blaster.o mailbox.o
make[1]: Leaving directory '/root/pi-blaster'

whiskthecat, 2023-01-07:
Surely some dependency is missing since this works fine on raspbian but not on dietpi

sarfata, 2023-01-09:
Can you see what is the status code after `pi-blaster` stops? Or run it with gdb? It might tell you more about why it stopped.

whiskthecat, 2023-01-17:
I manged to bork this SD card on accident, sorry. I am just going to use raspbian.

closed 2023-01-17 by whiskthecat