sarfata/pi-blaster #120: PWM output frequency is half of requested

whiskthecat, 2023-01-17:
Using the default settings I get 50Hz on my oscilloscope when pi-blaster thinks it is doing 100Hz.
If I recompile and change CYCLE_TIME_US to 5000 I get 100Hz but now pi-blaster reports this as 200Hz.
This is on a Raspberry Pi 3 Model B Plus Rev 1.3 running Raspbian.

```
#define CYCLE_TIME_US   10000
#define SAMPLE_US   10
#define NUM_SAMPLES (CYCLE_TIME_US/SAMPLE_US)
#define NUM_CBS     (NUM_SAMPLES*2)
```
```
ludy@ludylite:~/pi-blaster $ sudo pi-blaster
MBox Board Revision: 0xa020d3
DMA Channels Info: 0x7ff5, using DMA Channel: 14
Using hardware:                   PWM
Number of channels:                32
PWM frequency:                 100 Hz
PWM steps:                       1000
Maximum period (100  %):      10000us
Minimum period (0.100%):         10us
DMA Base:                  0x3f007000
Initialised, Daemonized, Reading /dev/pi-blaster.
```
```
ludy@ludylite:~/pi-blaster $ echo "17=0.5" > /dev/pi-blaster
```
[image]

sarfata, 2023-01-18:
Looks like we messed up some basic math. Should not be too hard to figure out where the issue is. Could you take a look at the source and suggest a fix in a pull request?

whiskthecat, 2023-01-19:
I honestly have no idea where to even begin. If you want me to test something though I can provide more oscope shots.