sarfata/pi-blaster #122: doesnt start

szampardi, 2023-10-27:
pretty much same as https://github.com/sarfata/pi-blaster/issues/119, only I'm on a fresh install of archlinuxarm on a pi zero 2w

PWM
```
Starting program: /usr/bin/pi-blaster -g12,13 -f
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
MBox Board Revision: 0x902120
DMA Channels Info: 0x7ff5, using DMA Channel: 14
Using hardware:                   PWM
Number of channels:                 2
PWM frequency:                 100 Hz
PWM steps:                       1000
Maximum period (100  %):      10000us
Minimum period (0.100%):         10us
DMA Base:                  0x20007000

Program received signal SIGBUS, Bus error.
0x0000007ff7e95b90 in memset () from /usr/lib/libc.so.6
```

PCM
```
Starting program: /usr/bin/pi-blaster -g12,13 -f -p
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
MBox Board Revision: 0x902120
DMA Channels Info: 0x7ff5, using DMA Channel: 14
Using hardware:                   PCM
Number of channels:                 2
PWM frequency:                 100 Hz
PWM steps:                       1000
Maximum period (100  %):      10000us
Minimum period (0.100%):         10us
DMA Base:                  0x20007000

Program received signal SIGBUS, Bus error.
0x0000007ff7e95b90 in memset () from /usr/lib/libc.so.6
```

PWM enabled in /boot/config.txt with `dtoverlay=pwm-2chan,pin=12,func=4,pin2=13,func2=4`

szampardi, 2023-10-27:
`Linux alarm 6.1.58-2-rpi-ARCH #1 SMP PREEMPT Fri Oct 20 14:01:47 MDT 2023 aarch64 GNU/Linux`

sarfata, 2023-10-28:
Thanks for firing up GDB and looking into this.

When it crashes, can you type `bt` to see what it was doing before the crash?

szampardi, 2023-10-28:
PWM
```
Reading symbols from pi-blaster...
(gdb) r
Starting program: /usr/bin/pi-blaster -f -g 18,19
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
MBox Board Revision: 0x902120
DMA Channels Info: 0x7ff5, using DMA Channel: 14
Using hardware:                   PWM
Number of channels:                 2
PWM frequency:                 100 Hz
PWM steps:                       1000
Maximum period (100  %):      10000us
Minimum period (0.100%):         10us
DMA Base:                  0x20007000

Program received signal SIGBUS, Bus error.
0x0000007ff7e95b90 in memset () from /usr/lib/libc.so.6
(gdb) bt
#0  0x0000007ff7e95b90 in memset () from /usr/lib/libc.so.6
#1  0x00000055555522bc in init_ctrl_data ()
#2  0x0000005555553348 in main ()
(gdb)
```

PCM
```
Reading symbols from pi-blaster...
(gdb) r
Starting program: /usr/bin/pi-blaster -f -g 18,19 -p
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
MBox Board Revision: 0x902120
DMA Channels Info: 0x7ff5, using DMA Channel: 14
Using hardware:                   PCM
Number of channels:                 2
PWM frequency:                 100 Hz
PWM steps:                       1000
Maximum period (100  %):      10000us
Minimum period (0.100%):         10us
DMA Base:                  0x20007000

Program received signal SIGBUS, Bus error.
0x0000007ff7e95b90 in memset () from /usr/lib/libc.so.6
(gdb) bt
#0  0x0000007ff7e95b90 in memset () from /usr/lib/libc.so.6
#1  0x00000055555522bc in init_ctrl_data ()
#2  0x0000005555553348 in main ()
(gdb)
```

sarfata, 2023-10-28:
Can you build with debugging symbols (edit `makefile.am`, add `-g -DDEBUG` to the end of the CFLAGS, and then re-build everything.

GDB should give you more details in the backtrace and a line number.

thanks

szampardi, 2023-10-28:
recompiled with `-g -rdynamic`
```
(gdb) bt
#0  0x0000007ff7e95b90 in memset () from /usr/lib/libc.so.6
#1  0x00000055555527bc in init_ctrl_data () at pi-blaster.c:681
#2  0x0000005555553848 in main (argc=5, argv=0x7ffffffbe8) at pi-blaster.c:1084
```

szampardi, 2023-10-28:
```
(gdb) b 667
Breakpoint 1 at 0x5555552708: file pi-blaster.c, line 668.
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /usr/local/src/pi-blaster/pi-blaster -f -g 18,19 -p
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
MBox Board Revision: 0x902120
DMA Channels Info: 0x7ff5, using DMA Channel: 14
Using hardware:                   PCM
Number of channels:                 2
PWM frequency:                 100 Hz
PWM steps:                       1000
Maximum period (100  %):      10000us
Minimum period (0.100%):         10us
DMA Base:                  0x20007000

Breakpoint 1, init_ctrl_data () at pi-blaster.c:668
668	  struct ctl *ctl = (struct ctl *)mbox.virt_addr;
(gdb) 
(gdb) 
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /usr/local/src/pi-blaster/pi-blaster -f -g 18,19 -p
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
MBox Board Revision: 0x902120
DMA Channels Info: 0x7ff5, using DMA Channel: 14
Using hardware:                   PCM
Number of channels:                 2
PWM frequency:                 100 Hz
PWM steps:                       1000
Maximum period (100  %):      10000us
Minimum period (0.100%):         10us
DMA Base:                  0x20007000

Breakpoint 1, init_ctrl_data () at pi-blaster.c:668
668	  struct ctl *ctl = (struct ctl *)mbox.virt_addr;
(gdb) step
669	  dma_cb_t *cbp = ctl->cb;
(gdb) step
671	  uint32_t phys_gpclr0 = GPIO_PHYS_BASE + 0x28;
(gdb) step
672	  uint32_t phys_gpset0 = GPIO_PHYS_BASE + 0x1c;
(gdb) 
675	  if (delay_hw == DELAY_VIA_PWM) {
(gdb) 
679	    phys_fifo_addr = PCM_PHYS_BASE + 0x04;
(gdb) 
681	  memset(ctl->sample, 0, sizeof(ctl->sample));
(gdb) 

Program received signal SIGBUS, Bus error.
0x0000007ff7e95b90 in memset () from /usr/lib/libc.so.6
(gdb)
```

szampardi, 2023-10-28:
```
Starting program: /usr/local/src/pi-blaster/pi-blaster -f -g 18,19 -p
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
MBox Request:
0000: 0x0000001c
0004: 0x00000000
0008: 0x00010002
000c: 0x00000004
0010: 0x00000000
0014: 0x00000000
0018: 0x00000000
MBox Response:
0000: 0x0000001c
0004: 0x80000000
0008: 0x00010002
000c: 0x00000004
0010: 0x80000004
0014: 0x00902120
0018: 0x00000000
MBox Board Revision: 0x902120
MBox Request:
0000: 0x0000001c
0004: 0x00000000
0008: 0x00060001
000c: 0x00000004
0010: 0x00000000
0014: 0x00000000
0018: 0x00000000
MBox Response:
0000: 0x0000001c
0004: 0x80000000
0008: 0x00060001
000c: 0x00000004
0010: 0x80000004
0014: 0x00007ff5
0018: 0x00000000
DMA Channels Info: 0x7ff5, using DMA Channel: 14
Using hardware:                   PCM
Number of channels:                 2
PWM frequency:                 100 Hz
PWM steps:                       1000
Maximum period (100  %):      10000us
Minimum period (0.100%):         10us
DMA Base:                  0x20007000
Requesting 69632 bytes
Alignment: 4096 bytes
mem_alloc flags:  0x1c
MBox Request:
0000: 0x00000024
0004: 0x00000000
0008: 0x0003000c
000c: 0x0000000c
0010: 0x0000000c
0014: 0x00011000
0018: 0x00001000
001c: 0x0000001c
0020: 0x00000000
MBox Response:
0000: 0x00000024
0004: 0x80000000
0008: 0x0003000c
000c: 0x0000000c
0010: 0x80000004
0014: 0x0000000a
0018: 0x00001000
001c: 0x0000001c
0020: 0x00000000
mem_ref 10
MBox Request:
0000: 0x0000001c
0004: 0x00000000
0008: 0x0003000d
000c: 0x00000004
0010: 0x00000004
0014: 0x0000000a
0018: 0x00000000
MBox Response:
0000: 0x0000001c
0004: 0x80000000
0008: 0x0003000d
000c: 0x00000004
0010: 0x80000004
0014: 0x9eb7d000
0018: 0x00000000
bus_addr = 0x9eb7d000
base=0x1eb7d000, mem=0x7ff7def000
virt_addr 0x7ff7def000
Initializing DMA ...

Program received signal SIGBUS, Bus error.
0x0000007ff7e95b90 in memset () from /usr/lib/libc.so.6
(gdb) 
```

sarfata, 2023-10-28:
great thanks

sarfata, 2023-10-28:
I don't see anything obviously wrong but I have not touched this code in a long time. I want to get back into it, I will report back if I have ideas but would definitely welcome help from anyone willing to spend time looking into this crash.

szampardi, 2023-10-28:
I'm sadly not experienced enough with C programming to help debug this deeper but am more than willing to test things out if you have anything to try, will stand by for updates.

Thanks!