sarfata/pi-blaster #125: debian bookworm

splitti, 2024-02-15:
Hey,

i tried a lot and searched also to get started with pi-blaster in bookworm 64bit version. There are a lot changes in the os. In 32bit it works, but i have to use it in 64bit mode.

Here are some errors:
` sudo dpkg -i pi-blaster_0.1.2_arm64.deb pi-blaster-dbgsym_0.1.2_arm64.deb
Vormals nicht ausgewähltes Paket pi-blaster wird gewählt.
(Lese Datenbank ... 93037 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von pi-blaster_0.1.2_arm64.deb ...
Entpacken von pi-blaster (0.1.2) ...
Vormals nicht ausgewähltes Paket pi-blaster-dbgsym wird gewählt.
Vorbereitung zum Entpacken von pi-blaster-dbgsym_0.1.2_arm64.deb ...
Entpacken von pi-blaster-dbgsym (0.1.2) ...
pi-blaster (0.1.2) wird eingerichtet ...
Job for pi-blaster.service failed because the control process exited with error code.
See "systemctl status pi-blaster.service" and "journalctl -xeu pi-blaster.service" for details.
invoke-rc.d: initscript pi-blaster, action "start" failed.
× pi-blaster.service - Daemon for PWM control of the Raspberry Pi GPIO pins
     Loaded: loaded (/lib/systemd/system/pi-blaster.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Thu 2024-02-15 01:33:34 CET; 28ms ago
    Process: 6771 ExecStart=/usr/sbin/pi-blaster $DAEMON_OPTS (code=exited, status=1/FAILURE)
        CPU: 5ms

Feb 15 01:33:34 MuPiBox pi-blaster[6771]: Number of channels:                 1
Feb 15 01:33:34 MuPiBox pi-blaster[6771]: PWM frequency:                 100 Hz
Feb 15 01:33:34 MuPiBox pi-blaster[6771]: PWM steps:                       1000
Feb 15 01:33:34 MuPiBox pi-blaster[6771]: Maximum period (100  %):      10000us
Feb 15 01:33:34 MuPiBox pi-blaster[6771]: Minimum period (0.100%):         10us
Feb 15 01:33:34 MuPiBox pi-blaster[6771]: DMA Base:                  0xfe007000
Feb 15 01:33:34 MuPiBox pi-blaster[6771]: pi-blaster stopped.
Feb 15 01:33:34 MuPiBox systemd[1]: pi-blaster.service: Control process exited, code=exited, status=1/FAILURE
Feb 15 01:33:34 MuPiBox systemd[1]: pi-blaster.service: Failed with result 'exit-code'.
Feb 15 01:33:34 MuPiBox systemd[1]: Failed to start pi-blaster.service - Daemon for PWM control of the Raspberry Pi GPIO pins.
dpkg: Fehler beim Bearbeiten des Paketes pi-blaster (--install):
 »installiertes post-installation-Skript des Paketes pi-blaster«-Unterprozess gab den Fehlerwert 1 zurück
dpkg: Abhängigkeitsprobleme verhindern Konfiguration von pi-blaster-dbgsym:
 pi-blaster-dbgsym hängt ab von pi-blaster (= 0.1.2); aber:
  Paket pi-blaster ist noch nicht konfiguriert.

dpkg: Fehler beim Bearbeiten des Paketes pi-blaster-dbgsym (--install):
 Abhängigkeitsprobleme - verbleibt unkonfiguriert
Trigger für man-db (2.11.2-2) werden verarbeitet ...
Fehler traten auf beim Bearbeiten von:
 pi-blaster
 pi-blaster-dbgsym
`

`sudo dpkg -i pi-blaster_0.1.2_arm64.deb
(Lese Datenbank ... 93050 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von pi-blaster_0.1.2_arm64.deb ...
Entpacken von pi-blaster (0.1.2) über (0.1.2) ...
pi-blaster (0.1.2) wird eingerichtet ...
Job for pi-blaster.service failed because the control process exited with error code.
See "systemctl status pi-blaster.service" and "journalctl -xeu pi-blaster.service" for details.
invoke-rc.d: initscript pi-blaster, action "start" failed.
× pi-blaster.service - Daemon for PWM control of the Raspberry Pi GPIO pins
     Loaded: loaded (/lib/systemd/system/pi-blaster.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Thu 2024-02-15 01:37:04 CET; 31ms ago
    Process: 18344 ExecStart=/usr/sbin/pi-blaster $DAEMON_OPTS (code=exited, status=1/FAILURE)
        CPU: 5ms

Feb 15 01:37:04 MuPiBox pi-blaster[18344]: Number of channels:                 1
Feb 15 01:37:04 MuPiBox pi-blaster[18344]: PWM frequency:                 100 Hz
Feb 15 01:37:04 MuPiBox pi-blaster[18344]: PWM steps:                       1000
Feb 15 01:37:04 MuPiBox pi-blaster[18344]: Maximum period (100  %):      10000us
Feb 15 01:37:04 MuPiBox pi-blaster[18344]: Minimum period (0.100%):         10us
Feb 15 01:37:04 MuPiBox pi-blaster[18344]: DMA Base:                  0xfe007000
Feb 15 01:37:04 MuPiBox pi-blaster[18344]: pi-blaster stopped.
Feb 15 01:37:04 MuPiBox systemd[1]: pi-blaster.service: Control process exited, code=exited, status=1/FAILURE
Feb 15 01:37:04 MuPiBox systemd[1]: pi-blaster.service: Failed with result 'exit-code'.
Feb 15 01:37:04 MuPiBox systemd[1]: Failed to start pi-blaster.service - Daemon for PWM control of the Raspberry Pi GPIO pins.
dpkg: Fehler beim Bearbeiten des Paketes pi-blaster (--install):
 »installiertes post-installation-Skript des Paketes pi-blaster«-Unterprozess gab den Fehlerwert 1 zurück
Trigger für man-db (2.11.2-2) werden verarbeitet ...
Fehler traten auf beim Bearbeiten von:
 pi-blaster
`

splitti, 2024-02-16:
# autogen.sh
```sudo ./autogen.sh
Generating configure files... may take a while.
configure.ac:42: installing './compile'
configure.ac:59: installing './config.guess'
configure.ac:59: installing './config.sub'
configure.ac:8: installing './install-sh'
configure.ac:8: installing './missing'
Makefile.am: installing './depcomp'
Preparing was successful if there was no error messages above.
Now type:
  ./configure && make
Run './configure --help' for more information
```
# configure
```sudo ./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... gcc3
checking whether make sets $(MAKE)... (cached) yes
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for strings.h... yes
checking for sys/stat.h... yes
checking for sys/types.h... yes
checking for unistd.h... yes
checking for sys/param.h... yes
checking for fcntl.h... yes
checking for stdint.h... (cached) yes
checking for stdlib.h... (cached) yes
checking for string.h... (cached) yes
checking for sys/ioctl.h... yes
checking for sys/time.h... yes
checking for unistd.h... (cached) yes
checking for _Bool... yes
checking for stdbool.h that conforms to C99... yes
checking for size_t... yes
checking for uint32_t... yes
checking for uint8_t... yes
checking for sys/mkdev.h... no
checking for sys/sysmacros.h... yes
checking build system type... aarch64-unknown-linux-gnu
checking host system type... aarch64-unknown-linux-gnu
checking for getpagesize... yes
checking for working mmap... yes
checking for memset... yes
checking for mkfifo... yes
checking for munmap... yes
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating config.h
config.status: executing depfiles commands
```
# make
````sudo make
make  all-am
make[1]: Verzeichnis „/home/dietpi/pi-blaster“ wird betreten
gcc -DHAVE_CONFIG_H -I.     -Wall -pedantic -std=c99 -MT pi-blaster.o -MD -MP -MF .deps/pi-blaster.Tpo -c -o pi-blaster.o pi-blaster.c
mv -f .deps/pi-blaster.Tpo .deps/pi-blaster.Po
gcc -DHAVE_CONFIG_H -I.     -Wall -pedantic -std=c99 -MT mailbox.o -MD -MP -MF .deps/mailbox.Tpo -c -o mailbox.o mailbox.c
mv -f .deps/mailbox.Tpo .deps/mailbox.Po
gcc  -Wall -pedantic -std=c99   -o pi-blaster pi-blaster.o mailbox.o
make[1]: Verzeichnis „/home/dietpi/pi-blaster“ wird verlassen
```

Service
```sudo service pi-blaster status
● pi-blaster.service - LSB: Daemon for PWM control of the Raspberry Pi GPIO pins
     Loaded: loaded (/etc/init.d/pi-blaster; generated)
     Active: active (exited) since Fri 2024-02-16 13:22:45 CET; 2h 1min ago
       Docs: man:systemd-sysv-generator(8)
    Process: 356 ExecStart=/etc/init.d/pi-blaster start (code=exited, status=0/SUCCESS)
        CPU: 4ms

Notice: journal has been rotated since unit was started, output may be incomplete.
dietpi@MuPiTestBox ~ $ sudo service stop
stop: unrecognized service
dietpi@MuPiTestBox ~ $ sudo service pi-blaster restart
dietpi@MuPiTestBox ~ $ sudo service pi-blaster status
● pi-blaster.service - LSB: Daemon for PWM control of the Raspberry Pi GPIO pins
     Loaded: loaded (/etc/init.d/pi-blaster; generated)
     Active: active (exited) since Fri 2024-02-16 15:24:06 CET; 2s ago
       Docs: man:systemd-sysv-generator(8)
    Process: 396255 ExecStart=/etc/init.d/pi-blaster start (code=exited, status=0/SUCCESS)
        CPU: 4ms

Feb 16 15:24:06 MuPiTestBox systemd[1]: Starting pi-blaster.service - LSB: Daemon for PWM control of the Raspberry Pi GPIO pins...
Feb 16 15:24:06 MuPiTestBox systemd[1]: Started pi-blaster.service - LSB: Daemon for PWM control of the Raspberry Pi GPIO pins.
```

splitti, 2024-02-22:
Here is the config.log
```creating ./config.status

## ---------------------- ##
## Running config.status. ##
## ---------------------- ##

This file was extended by pi-blaster config.status 0.1.3, which was
generated by GNU Autoconf 2.71.  Invocation command line was

  CONFIG_FILES    =
  CONFIG_HEADERS  =
  CONFIG_LINKS    =
  CONFIG_COMMANDS =
  $ ./config.status

on FelixBox

config.status:879: creating Makefile
config.status:879: creating config.h
config.status:1108: executing depfiles commands
config.status:1185: cd .       && sed -e '/# am--include-marker/d' Makefile         | make -f - am--depfiles
config.status:1190: $? = 0

## ---------------- ##
## Cache variables. ##
## ---------------- ##

ac_cv_build=aarch64-unknown-linux-gnu
ac_cv_c_compiler_gnu=yes
ac_cv_c_uint32_t=yes
ac_cv_c_uint8_t=yes
ac_cv_env_CC_set=
ac_cv_env_CC_value=
ac_cv_env_CFLAGS_set=
ac_cv_env_CFLAGS_value=
ac_cv_env_CPPFLAGS_set=
ac_cv_env_CPPFLAGS_value=
ac_cv_env_LDFLAGS_set=
ac_cv_env_LDFLAGS_value=
ac_cv_env_LIBS_set=
ac_cv_env_LIBS_value=
ac_cv_env_PKG_CONFIG_LIBDIR_set=
ac_cv_env_PKG_CONFIG_LIBDIR_value=
ac_cv_env_PKG_CONFIG_PATH_set=
ac_cv_env_PKG_CONFIG_PATH_value=
ac_cv_env_PKG_CONFIG_set=
ac_cv_env_PKG_CONFIG_value=
ac_cv_env_build_alias_set=
ac_cv_env_build_alias_value=
ac_cv_env_host_alias_set=
ac_cv_env_host_alias_value=
ac_cv_env_target_alias_set=
ac_cv_env_target_alias_value=
ac_cv_func_getpagesize=yes
ac_cv_func_memset=yes
ac_cv_func_mkfifo=yes
ac_cv_func_mmap_fixed_mapped=yes
ac_cv_func_munmap=yes
ac_cv_header_fcntl_h=yes
ac_cv_header_inttypes_h=yes
ac_cv_header_stdbool_h=yes
ac_cv_header_stdint_h=yes
ac_cv_header_stdio_h=yes
ac_cv_header_stdlib_h=yes
ac_cv_header_string_h=yes
ac_cv_header_strings_h=yes
ac_cv_header_sys_ioctl_h=yes
ac_cv_header_sys_mkdev_h=no
ac_cv_header_sys_param_h=yes
[... 151 lines cut ...]
#define HAVE_INTTYPES_H 1
#define HAVE_STDINT_H 1
#define HAVE_STRINGS_H 1
#define HAVE_SYS_STAT_H 1
#define HAVE_SYS_TYPES_H 1
#define HAVE_UNISTD_H 1
#define HAVE_SYS_PARAM_H 1
#define STDC_HEADERS 1
#define HAVE_FCNTL_H 1
#define HAVE_STDINT_H 1
#define HAVE_STDLIB_H 1
#define HAVE_STRING_H 1
#define HAVE_SYS_IOCTL_H 1
#define HAVE_SYS_TIME_H 1
#define HAVE_UNISTD_H 1
#define HAVE__BOOL 1
#define MAJOR_IN_SYSMACROS 1
#define HAVE_GETPAGESIZE 1
#define HAVE_MMAP 1
#define HAVE_MEMSET 1
#define HAVE_MKFIFO 1
#define HAVE_MUNMAP 1

configure: exit 0
```

m0yellow, 2025-02-16:
I can reproduce the crash: 
```
$ sudo gdb /usr/sbin/pi-blaster -f
GNU gdb (Debian 13.1-3) 13.1
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/sbin/pi-blaster...
Reading symbols from /usr/lib/debug/.build-id/85/0b601d0d2b46847a7f3da11d0025913d4cf4dd.debug...
(No debugging symbols found in /usr/lib/debug/.build-id/85/0b601d0d2b46847a7f3da11d0025913d4cf4dd.debug)
(gdb) run
Starting program: /usr/sbin/pi-blaster 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
MBox Board Revision: 0x902120
DMA Channels Info: 0x7ff5, using DMA Channel: 14
Using hardware:                   PWM
Number of channels:                32
PWM frequency:                 100 Hz
PWM steps:                       1000
Maximum period (100  %):      10000us
Minimum period (0.100%):         10us
DMA Base:                  0x20007000

Program received signal SIGBUS, Bus error.
__memset_zva64 () at ../sysdeps/aarch64/multiarch/../memset.S:126
126	../sysdeps/aarch64/multiarch/../memset.S: No such file or directory.
(gdb) bt
#0  __memset_zva64 () at ../sysdeps/aarch64/multiarch/../memset.S:126
#1  0x000000555555216c in init_ctrl_data ()
#2  0x00000055555531b0 in main ()
```
does this help? This is a fresh bookwork piZero2W out of the box, installed today, then apt-get update && apt-get full-upgrade,
```
sudo apt-get install debhelper dh-autoreconf dpkg-dev init-system-helpers autoconf dh-virtualenv git 
```
after cloning I did ...
```
dpkg-buildpackage -us -uc -i && sudo dpkg -i ../pi-blaster*.deb
```
I have to admit, I forgot about 64bit, just wanted to get started with pi-blaster again (after changing HW).
If it helps this pi is free to experiment with, and can be set online 24/7 with little effort.