sarfata/pi-blaster #126: Failed to start pi-blaster.service - Daemon for PWM control of the Raspberry Pi GPIO pins

turokhan, 2024-02-23:
Hi,

Installed as per instructions, but it doesn't work.

Failed to start pi-blaster.service - Daemon for PWM control of the Raspberry Pi GPIO pins

Docker install also doesnt work.

Unable to find image 'sarfata/pi-blaster:latest' locally

Is there an alternative?

sarfata, 2024-02-27:
Not 💀 - [We merged a PR two weeks ago.](https://github.com/sarfata/pi-blaster/pull/124) 

[image]

sarfata, 2024-02-27:
> Installed as per instructions, but it doesn't work.

You really need to send more details :)

sarfata, 2024-03-05:
@thalesfsp Why don't you rebuild the image locally? The Readme includes instructions:
> ```
> docker build -t pib .
> docker run -it --privileged --rm -v /dev:/dev pib
> ```