sarfata/pi-blaster #128: Pi-Blaster  Raspberry PI3+ Bullseye : brushless motors do not run

peniblou, 2024-03-29:
Bonjour Pi-Blaster team.  

I have a Raspberry PI 3+.     It was running perfectly with ServoBlaster and Stretch. 

I had to install Bullseye, and took advantage of this new environment to install Pi-Blaster (replacing Servo-Blaster).   Installation was easy and successful. However, I cannot make it work...

 ESC is waiting for a signal between 1 ms and 2 ms.   I did :
pi@raspberrypi: /pi-blaster$ echo "4=0.1" >  /dev/pi-blaster       (expecting a 1ms signal) 
pi@raspberrypi: /pi-blaster$ echo "4=0.11" >  /dev/pi-blaster      (expecting a 1.1 ms signal)
But no noise , nothing ...

After few (multiple ) tries I did :
pi@raspberrypi: /pi-blaster$ echo "4=0.01" >  /dev/pi-blaster   and          
pi@raspberrypi: /pi-blaster$ echo "4=0.011" >  /dev/pi-blaster   
The brushless motor then works very badly : forward and backward every seconds.   No other values are working 0.012 0.009...
Remark: I though it was a problem of frequency ( ESC is waiting for 20ms signal = 50hz when Pi-Blaster mentions 100 Hz) but 
the problem 120 issued by Whiskthecat specifies that we have a 50 Hz signal ...)

So my questions are:
     - Is my environment Raspberry PI 3+  /  Bullseye compatible with Pi-Blaster ?
     - Did I do something wrong ? Can you confirm that my first command  ( value =0.1 and 0.11 ) should work ? 
     - If you do not see any possible fix,  can I just use Servo Blaster   (and uninstall Pi-Blaster)  ? 

Many thanks for your help and support 

Michel Hirvois

sarfata, 2024-04-08:
Hi Michel,

The rpi 3+ should work well.

Are you able to hook an oscilloscope to the raspberry pi to see what is actually generated?

Can you copy/paste the text information displayed by `pi-blaster` when you start it?

> If you do not see any possible fix, can I just use Servo Blaster (and uninstall Pi-Blaster) ?

Yes of course. I am not sure I understand why that would not be an option.

peniblou, 2024-04-09:
Bonjour Thomas,

Thanks for coming back to me. 
Unfortunately, I do not have an oscilloscope to check the signal generated by the pi. 

I have done some additional testing.  To ease things, I will ask 2 questions 

- Q1- expected parameters for an ESC motor brushless
I need a 50hz signal. The pulse signal should vary from 1ms to 2ms ..   See figure below. 
What parameters do you recommend ? 
[image]

-Q2- I am not sure of the procedure to change the parameters ... 
This is what I do
       o I edit the pi-blaster.c file and change the parameters cycle_time_us ( 2000) and sample_us (100)...
       o make pi-blaster
          pi@raspberrypi:~/pi-blaster $ make pi-blaster
          gcc  -Wall -pedantic -std=c99   -o pi-blaster pi-blaster.o mailbox.o_
       o Check new parameters 
           pi@raspberrypi:~/pi-blaster $ ./pi-blaster
           MBox Board Revision: 0xa52082
          DMA Channels Info: 0x7ff5, using DMA Channel: 14
          Using hardware:                   PWM
         Number of channels:                32
         PWM frequency:                  50 Hz
         PWM steps:                         20
         Maximum period (100  %):       2000us
         Minimum period (5.000%):        100us
         DMA Base:                  0x3f007000
         pi-blaster: Failed to open /dev/mem: Permission denied
         pi-blaster stopped.

I am not sure that my changes are taken into account. ( I am not an expert in C and compilation... )  Even if I delete the files pi-blaster.c et pi-blaster ,  pi-blaster still works  ! (badly).    For information, my pi-blaster starts automatically at boot. 

Many thanks in advance for your help

Michel Hirvois

peniblou, 2024-04-22:
Bonjour ,

  I have installed ServoBlaster with still some issue that I have added in[ ](https://github.com/richardghirst/PiBits/issues/132)

Therefore I close this current issue #128

closed 2024-04-22 by peniblou